muka/go-bluetooth #2: Example code does not work any more

mrfuxi, 2017-05-09:
```
$ go run watch_changes.go 
# command-line-arguments
./watch_changes.go:84: cannot use func literal (type func(emitter.Event)) as type *emitter.Callback in argument to emitter.On
./watch_changes.go:115: cannot use func literal (type func(emitter.Event)) as type *emitter.Callback in argument to api.On
./watch_changes.go:170: cannot use func literal (type func(emitter.Event)) as type *emitter.Callback in argument to dev.On
./watch_changes.go:232: undefined: api.RefreshManagerState
./watch_changes.go:240: undefined: err
./watch_changes.go:241: undefined: err
./watch_changes.go:242: undefined: err
```

One of the problems I come across when trying code from examples

closed 2017-05-24 by muka

muka, 2017-05-24:
Should work now, let me know. Thanks

mrfuxi, 2017-05-24:
Thanks for the fixes!
I had to change package back to main to make it run. Any idea how to run it other way?
It panics when there are no args because of: `os.Args[1]`

```
panic: runtime error: index out of range

goroutine 1 [running]:
panic(0x65b6a0, 0xc82000a100)
	/usr/lib/go-1.6/src/runtime/panic.go:481 +0x3e6
main.main()
	/path/examples.go:13 +0x1fd
exit status 2
```

muka, 2017-05-24:
I see, definitely I should take more care to the examples.. will review them soon

It was supposed to work with an argument like `watch` but miss a nil check

Even better I would suggest to copy one of the examples to your own project and edit the code directly

commit c23d2e9c (2017-05-24): refactor examples fixes #2 #3