muka/go-bluetooth #4: getting nil in device properties

saurabh-newera, 2017-06-16:
i am getting run time error while discovering sensors.....

2017/06/16 03:51:41 Loaded devices %d [{/org/bluez/hci0/dev_B0_B4_48_D2_30_85 <nil> 0x106d9160 map[]}]
2017/06/16 03:51:41 device:  {/org/bluez/hci0/dev_B0_B4_48_D2_30_85 <nil> 0x106d9160 map[]}
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x180a04]

goroutine 3 [running]:
main.filterDevice(0x10639f2c, 0x106e61b0)
        /home/Embs1PSL/github/DistributedEdgeIoT/src/securiot-ble/watch_changes.go:166 +0x2c
main.loadDevices(0x10670900)
        /home/Embs1PSL/github/DistributedEdgeIoT/src/securiot-ble/watch_changes.go:156 +0x2d8
main.deviceIsCached(0x2b49d0)
        /home/Embs1PSL/github/DistributedEdgeIoT/src/securiot-ble/watch_changes.go:140 +0x94
main.discoverDevices(0x1d8679, 0x4)
        /home/Embs1PSL/github/DistributedEdgeIoT/src/securiot-ble/watch_changes.go:112 +0x14
main.waitAdapter()
        /home/Embs1PSL/github/DistributedEdgeIoT/src/securiot-ble/watch_changes.go:103 +0x274
created by main.discoverDevice
        /home/Embs1PSL/github/DistributedEdgeIoT/src/securiot-ble/watch_changes.go:74 +0x54
exit status 2

muka, 2017-06-16:
Can you provide some more details: (version of bluez, type of BLE device connected)

Looking at `filterDevice` I can guess `Properties` is empty, is it correct?
https://github.com/muka/go-bluetooth/blob/master/examples/watch_changes.go#L148

saurabh-newera, 2017-06-17:
I am using raspberry pi with bluez version  5.23 and i am using TI Sensortags.
yes i am getting empty property of device and this is the issue.
I am executing the code given in the example...for getting device property
device := profile.NewDevice1(
		fmt.Sprintf(
			"/org/bluez/%s/dev_%s",
			adapterID,
			strings.Replace(deviceID, ":", "_", -1),
		),
	)

	log.Info("Device info\n---")
	log.Infof("Name: %s\n", device.Properties.Name)
	log.Infof("Modalias: %s\n", device.Properties.Modalias)

output:- 
---
2017/06/17 03:51:38 Name:
2017/06/17 03:51:38 Modalias:
2017/06/17 03:51:38 [{/org/bluez/hci0/dev_B0_B4_48_D2_30_85 <nil> 0x106d9960 map[]}]

you can see i am getting empty name and other properties empty too. why?

Thank you.

muka, 2017-06-18:
Bluez should be upgraded to v5.43, there are some instructions in the readme

mholt, 2017-08-10:
Hi, this is pretty cool so far; but when I used your script to update bluez I still have nil Properties (Ubuntu 16.04). `bluetoothd -v` shows 5.43. Didn't see any errors. Did I miss something?

muka, 2017-08-11:
Hi, can you provide more details, like which kind of device are you interfacing and code to use so I can try to reproduce

Thanks!

mholt, 2017-08-11:
Hey @muka, thanks for responding! (Yours is the only lib I could find that even remotely worked for what I need.)

Here's what I did, log is below:
I ran `./bluez-install.sh` - it looks like it succeeded. Full output is below. Looks like it succeeded (but I may be wrong/overlooking something?) - output is long so I've collapsed it; click to expand:

<details>
```
$ bash ./bluez-install.sh 
Hit:1 http://us.archive.ubuntu.com/ubuntu xenial InRelease
Get:2 http://us.archive.ubuntu.com/ubuntu xenial-updates InRelease [102 kB]
Get:3 http://us.archive.ubuntu.com/ubuntu xenial-backports InRelease [102 kB]
Get:4 http://us.archive.ubuntu.com/ubuntu xenial-updates/main amd64 Packages [608 kB]                         
Get:5 http://us.archive.ubuntu.com/ubuntu xenial-updates/main i386 Packages [586 kB]                                         
Get:6 http://us.archive.ubuntu.com/ubuntu xenial-updates/main amd64 DEP-11 Metadata [305 kB]                                 
Get:7 http://us.archive.ubuntu.com/ubuntu xenial-updates/main DEP-11 64x64 Icons [207 kB]                                    
Get:8 http://us.archive.ubuntu.com/ubuntu xenial-updates/universe amd64 Packages [512 kB]                                    
Get:9 http://us.archive.ubuntu.com/ubuntu xenial-updates/universe i386 Packages [495 kB]                                     
Ign:10 http://dl.google.com/linux/chrome/deb stable InRelease                                                                
Get:11 http://us.archive.ubuntu.com/ubuntu xenial-updates/universe amd64 DEP-11 Metadata [170 kB]                            
Get:12 http://us.archive.ubuntu.com/ubuntu xenial-updates/universe DEP-11 64x64 Icons [209 kB]                               
Get:13 http://us.archive.ubuntu.com/ubuntu xenial-updates/multiverse amd64 DEP-11 Metadata [5,888 B]                         
Get:14 http://us.archive.ubuntu.com/ubuntu xenial-backports/main amd64 DEP-11 Metadata [3,324 B]                             
Get:15 http://us.archive.ubuntu.com/ubuntu xenial-backports/universe amd64 DEP-11 Metadata [5,136 B]                         
Get:16 http://dl.google.com/linux/chrome/deb stable Release [1,189 B]                                                        
Get:17 http://dl.google.com/linux/chrome/deb stable Release.gpg [819 B]          
Get:18 http://security.ubuntu.com/ubuntu xenial-security InRelease [102 kB]              
Get:19 http://dl.google.com/linux/chrome/deb stable/main amd64 Packages [1,372 B]                
Get:20 http://security.ubuntu.com/ubuntu xenial-security/main amd64 Packages [340 kB]         
Get:21 http://security.ubuntu.com/ubuntu xenial-security/main i386 Packages [319 kB]
Get:22 http://security.ubuntu.com/ubuntu xenial-security/main Translation-en [143 kB]
Get:23 http://security.ubuntu.com/ubuntu xenial-security/main amd64 DEP-11 Metadata [59.9 kB]
Get:24 http://security.ubuntu.com/ubuntu xenial-security/main DEP-11 64x64 Icons [52.0 kB]
Get:25 http://security.ubuntu.com/ubuntu xenial-security/universe amd64 Packages [153 kB]
Get:26 http://security.ubuntu.com/ubuntu xenial-security/universe i386 Packages [135 kB]
Get:27 http://security.ubuntu.com/ubuntu xenial-security/universe Translation-en [78.6 kB]
Get:28 http://security.ubuntu.com/ubuntu xenial-security/universe amd64 DEP-11 Metadata [48.7 kB]
Get:29 http://security.ubuntu.com/ubuntu xenial-security/universe DEP-11 64x64 Icons [64.2 kB]
Fetched 4,809 kB in 1s (3,026 kB/s)                                   
Reading package lists... Done
(Reading database ... 209714 files and directories currently installed.)
Preparing to unpack .../udev_229-4ubuntu19_amd64.deb ...
Unpacking udev (229-4ubuntu19) over (229-4ubuntu16) ...
Preparing to unpack .../libudev1_229-4ubuntu19_amd64.deb ...
Unpacking libudev1:amd64 (229-4ubuntu19) over (229-4ubuntu16) ...
Processing triggers for ureadahead (0.100.0-19) ...
Processing triggers for systemd (229-4ubuntu19) ...
Processing triggers for man-db (2.7.5-1) ...
Processing triggers for libc-bin (2.23-0ubuntu5) ...
Setting up libudev1:amd64 (229-4ubuntu19) ...
Processing triggers for libc-bin (2.23-0ubuntu5) ...
Selecting previously unselected package libudev-dev:amd64.
(Reading database ... 209714 files and directories currently installed.)
Preparing to unpack .../libudev-dev_229-4ubuntu19_amd64.deb ...
Unpacking libudev-dev:amd64 (229-4ubuntu19) ...
Preparing to unpack .../wget_1.17.1-1ubuntu1.2_amd64.deb ...
Unpacking wget (1.17.1-1ubuntu1.2) over (1.17.1-1ubuntu1.1) ...
Selecting previously unselected package libdbus-1-dev:amd64.
Preparing to unpack .../libdbus-1-dev_1.10.6-1ubuntu3.3_amd64.deb ...
Unpacking libdbus-1-dev:amd64 (1.10.6-1ubuntu3.3) ...
Selecting previously unselected package libdbus-glib-1-dev.
Preparing to unpack .../libdbus-glib-1-dev_0.106-1_amd64.deb ...
Unpacking libdbus-glib-1-dev (0.106-1) ...
Selecting previously unselected package libical-dev:amd64.
Preparing to unpack .../libical-dev_1.0.1-0ubuntu2_amd64.deb ...
Unpacking libical-dev:amd64 (1.0.1-0ubuntu2) ...
Selecting previously unselected package libtinfo-dev:amd64.
Preparing to unpack .../libtinfo-dev_6.0+20160213-1ubuntu1_amd64.deb ...
Unpacking libtinfo-dev:amd64 (6.0+20160213-1ubuntu1) ...
Selecting previously unselected package libreadline6-dev:amd64.
Preparing to unpack .../libreadline6-dev_6.3-8ubuntu2_amd64.deb ...
[... 1475 lines cut ...]
```
$ sudo mv $(which bluetoothd) $(which bluetoothd)_bak
[... cut ...]
Aha, okay, now it should be 5.43, right? I don't know, I probably am doing the wrong thing.

```
$ sudo systemctl daemon-reload
$ sudo service bluetooth restart
$ sudo hciconfig hci0 up
```

Trying again:

```
2017/08/10 17:07:00 Started discovery
discovery: &api.Device{Path:"/org/bluez/hci0/dev_**_**_**_**_**_**", Properties:(*profile.Device1Properties)(nil), client:(*profile.Device1)(0xc42000d120), chars:map[dbus.ObjectPath]*profile.GattCharacteristic1(nil)}
```

Properties field is still nil.

Any idea what I did wrong? :sweat_smile:

muka, 2017-08-19:
Hey,
sorry for the delay. Thank you for the detailed report!

On the install side it seems ok .. 
Can you try to add the `-E` (experimental) flag to bluez? 
You should find it in its systemd service file (somewhere under `/etc/systemd/**`) and restart the service
In that way all the new GATT related features are avail, it may be missing in the docs

If you are using the watch changes example with no changes I guess it may not match the proper characteristic and services so may be a cause. 

I would suggest to install the `d-feet` package to view the dbus exposed services. It will show you a list of interfaces, including the `bluez` one with all the devices found listed.
On ubuntu you can use the bluetooth UI to scan for devices, afterwards the list in d-feet will be updated

Let me know

mholt, 2017-08-24:
Thanks Muka. The only machine I have to test this on is already being used for something in our lab, so I'll have to wait until I have a chance to hack on it again to make it work. I'm not sure when that'll be. Sorry... We can close this until I have an opportunity to come back to this.

muka, 2017-08-24:
Ok, thanks anyway. Best

closed 2017-08-24 by muka