muka/go-bluetooth #5: Cannot find TemperatureConfig characteristic

saurabh-newera, 2017-06-17:
I am trying to get temperature data form TI sensorTag with the help of tagAddress but i am facing the issue mentioned below.

var tagAddress = "B0:B4:48:D2:30:85"
dev, err := api.GetDeviceByAddress(tagAddress)
	if err != nil {
		panic(err)
	}
       log.Debug("dev ",dev)
	
	if dev == nil {
		panic("Device not found")
	}
err = dev.Connect()
	if err != nil {
		panic(err)
	}
sensorTag, err := devices.NewSensorTag(dev)
	if err != nil {
		panic(err)
	}

	log.Debug("sensorTag: ",sensorTag)
	
	var notifyTemperature = func(fn func(temperature float64)) {
		sensorTag.Temperature.StartNotify()
		select {}
	}
	
	notifyTemperature(func(t float64) {
		log.Debug("Temperature update: %f", t)
	})

OUTPUT:- 

2017/06/17 14:32:17 dev &{/org/bluez/hci0/dev_B0_B4_48_D2_30_85 0x10696e40 0x106e70f0 map[]}
panic: Cannot find TemperatureConfig characteristic F000AA02-0451-4000-B000-000000000000

goroutine 1 [running]:
main.SensorTagTemperatureExample()
        /home/Embs1PSL/github/DistributedEdgeIoT/src/securiot-ble/sensortag_temperature.go:52 +0x3bc
main.main()
        /home/Embs1PSL/github/DistributedEdgeIoT/src/securiot-ble/sensortag_temperature.go:19 +0x14
exit status 2

muka, 2017-06-19:
See #4 in more recent version of Bluez the support for GATT profiles improved a lot and had no option to support backward compatibility. Let me know if it solve your issue.

saurabh-newera, 2017-06-19:
After updating bluez to version 5.43 i am getting temperature data. But i think the example that you have given for fetching sensorTag temperature  data and the library code are out of sync.

when i was using the code given in example for fetching temperature data, for that i was not getting data in callback. 
example code- sensorTag_temperature
var notifyTemperature = func(fn func(temperature float64)) {
		sensorTag.Temperature.StartNotify()
		select {}
	}
	notifyTemperature(func(t float64) {
		logger.Infof("Temperature update: %f", t)
	})
Please update me why i am not getting data in the callback. It just got stuck after calling  function notifyTemperature.

so i added - 
 dev.On("data", emitter.NewCallback(func(ev emitter.Event) {
			 log.Debug(ev)	
		}))
in ev now i am getting temperature data.
output:-
{/org/bluez/hci0/dev_24_71_89_BE_7F_81.data {0x10772540 temperature 27 C}}
{/org/bluez/hci0/dev_24_71_89_BE_7F_81.data {0x10772540 temperature 26.96875 C}}

muka, 2017-06-19:
Thank you for catching the bug, can you provide a PR with the fix ?

saurabh-newera, 2017-06-19:
sure, i can give you a PR with the fix.
I have few other doubts, can we chat on skype or whatsapp.

muka, 2017-07-07:
Closing in the meanwhile, let me know if you plan to provide a PR

closed 2017-07-07 by muka