muka/go-bluetooth #6: [Query] How to identify service/characteristics from events received

saurabh-newera, 2017-06-22:
Hi,
I have enhanced your code to read multiple sensor data  (e.g. humidity and Magnetometer/Accelerometer etc). I am getting following events for 2 services/characteristics : 

 Got update %v &{:1.3 /org/bluez/hci0/dev_A0_E6_F8_AD_23_04/service0037/char0038 org.freedesktop.DBus.Properties.PropertiesChanged [org.bluez.GattCharacteristic1 map[Value:@ay [0xe3, 0xfd, 0xa0, 0x2, 0x86, 0x0, 0x68, 0x2, 0xda, 0x1, 0x86, 0xef, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0]] []]}

 Got update %v &{:1.3 /org/bluez/hci0/dev_A0_E6_F8_AD_23_04/service0027/char0028 org.freedesktop.DBus.Properties.PropertiesChanged [org.bluez.GattCharacteristic1 map[Value:@ay [0xbe, 0xbe, 0xbe, 0xbe]] []]}

From these events, how do I identify  different sensors?

Are these dbus channel per sensor type?

muka, 2017-06-23:
There should be a `org.bluez.GattCharacteristic1` instance to look for in the service/characteristic list, is it the case?

muka, 2017-07-07:
Closing, please reopen if the problem persist

closed 2017-07-07 by muka