muka/go-bluetooth #8: BlueZ 5.46 error: No such field: NotifyAcquired in obj, No such field: WriteAcquired in obj

gbrayut, 2017-09-18:
It looks like there is an issue trying to use this library with the latest version of BlueZ. I used your script to upgrade to version 5.46 so that I could use the new [acquire-write and acquire-notify](https://unix.stackexchange.com/questions/392103) features. But now when I try using `api.GetDeviceByAddress(deviceID)` I get the following error messages:

```
root@GBRPI01:~/go/src/gitlab.com/gbrayut/rpi# go run cmd/ble-test/main.go
Starting BLE test app
hciconfig status: &linux.HCIConfigResult{Enabled:true, Address:"B8:27:EB:CF:0F:5B", Type:"Primary", Bus:"UART"}
panic: No such field: NotifyAcquired in obj

goroutine 1 [running]:
gitlab.com/gbrayut/rpi/vendor/github.com/muka/go-bluetooth/bluez/profile.NewGattCharacteristic1(0x11aea180, 0x3a, 0x11aea180)
        /home/gbray/go/src/gitlab.com/gbrayut/rpi/vendor/github.com/muka/go-bluetooth/bluez/profile/GattCharacteristic1.go:27 +0x1a4
gitlab.com/gbrayut/rpi/vendor/github.com/muka/go-bluetooth/api.(*Device).GetCharByUUID(0x11ab67c0, 0x11afaf90, 0x24, 0x0, 0x0, 0x70)
        /home/gbray/go/src/gitlab.com/gbrayut/rpi/vendor/github.com/muka/go-bluetooth/api/Device.go:256 +0x328
main.main()
        /home/gbray/go/src/gitlab.com/gbrayut/rpi/cmd/ble-test/main.go:59 +0x1a4
exit status 2

root@GBRPI01:~/go/src/gitlab.com/gbrayut/rpi# go run cmd/ble-test/main.go
Starting BLE test app
hciconfig status: &linux.HCIConfigResult{Enabled:true, Address:"B8:27:EB:CF:0F:5B", Type:"Primary", Bus:"UART"}
panic: No such field: WriteAcquired in obj

goroutine 1 [running]:
gitlab.com/gbrayut/rpi/vendor/github.com/muka/go-bluetooth/bluez/profile.NewGattCharacteristic1(0x1161c380, 0x3a, 0x1161c380)
        /home/gbray/go/src/gitlab.com/gbrayut/rpi/vendor/github.com/muka/go-bluetooth/bluez/profile/GattCharacteristic1.go:27 +0x1a4
gitlab.com/gbrayut/rpi/vendor/github.com/muka/go-bluetooth/api.(*Device).GetCharByUUID(0x11655360, 0x11616e40, 0x24, 0x0, 0x0, 0x70)
        /home/gbray/go/src/gitlab.com/gbrayut/rpi/vendor/github.com/muka/go-bluetooth/api/Device.go:256 +0x328
main.main()
        /home/gbray/go/src/gitlab.com/gbrayut/rpi/cmd/ble-test/main.go:59 +0x1a4
exit status 2
```

These seem to be from the new features in 5.46. I was able to add the two missing properties to GattCharacteristic1Properties in my vendor\github.com\muka\go-bluetooth\bluez\profile\GattCharacteristic1.go file and it started working again 

```
// GattCharacteristic1Properties exposed properties for GattCharacteristic1
type GattCharacteristic1Properties struct {
	Value          []byte
	Notifying      bool
	NotifyAcquired bool
	WriteAcquired  bool
	Service        dbus.ObjectPath
	UUID           string
	Flags          []string
	Descriptors    []dbus.ObjectPath
}
```

Not sure if there is a way to make it backwards compatible or not, but just thought I would give you a heads up.

muka, 2017-09-18:
Good find.. I am not able to support backward compatibility in term of effort and time, so I would go for the same modification you did and support the latest version.

Do you want to issue a PR for the fix you described?

Thanks

gbrayut, 2017-09-18:
Created a PR. If I find time this week I'll try to implement the Acquire-Write and Acquire-Notify calls. Using file descriptors to read/write from a Gatt Characteristic makes serial communication with the HM-10 BLE module much simpler.

muka, 2017-09-19:
Merged, thank you. 

Would be great if you could provide the implementation, I just have a TI SensorTag at hand for testing do not know if that is supported or required. Also the HM-10 BLE you mention is interesting, will try to buy one for testing.

I am currently working to integrate the "server" mode in the spare time. This hopefully should help to setup unit tests for both client and service/server side.

Best

muka, 2017-09-20:
Reopening to keep track of PR #10

muka, 2017-10-07:
Merged #10 example code is welcome

Thanks

PR #9 (2017-09-19): Add NotifyAcquired and WriteAcquired to bluez GattCharacteristic1Properties
Required for BlueZ 5.64 or greater

Fixes #8
commit 1c9f7e88 (2017-09-19): Add NotifyAcquired and WriteAcquired to bluez GattCharacteristic1Properties

Required for BlueZ 5.64 or greater

Fixes #8
closed 2017-10-07 by muka