muka/go-bluetooth #17: Any way to issue a "Powered" command?

dangerousHobo, 2017-10-09:
Hello,

I'm looking for a way to issue the "Powered" command (from the "Set Powered Command" section in mgmt-api.txt). Is there a way to issue a Powered on, Powered off with the current API?

Thanks.

muka, 2017-10-10:
Hi, I think this related to the hci command socket which is not implemented
as of now.

I was looking at those to enable device services from bluetoothd api

If you have material to work on I'd be happy to work on it

sent from my phone

dangerousHobo, 2017-10-10:
I'm using this package in a project for my employer. Currently mainly interested in the device properties, but it would be nice to issue commands like what is available via bluetoothctl and hcitool.

If you have anything I'd be happy to test it out on our embedded system (Linux 4.1, BlueZ 5.42, LSR TiWi5 wifi/bluetooth radio).

muka, 2017-10-10:
Thank you for your availability

Nothing available yet, you may run hci cli commands from go as done in
linux/hciconfig.go

Not optimal but helps to get things done.

The protocol to talk via hci socket need some study before implementing
anything, will put in my to-do list as it may enable other features also

sent from my phone

muka, 2017-10-13:
Opened #21 for progresses on the topic

closed 2018-01-13 by muka