muka/go-bluetooth #19: Question: OBEX D-Bus API implementation

BlackGh0st, 2017-10-11:
Hey there,

I'm currently working on a OBEX D-Bus API Client implementation in golang. (especially on the OOP profile) 

Since this project also uses the D-Bus to communicate with the D-Bus I was just wondering if this project would be the right place for such an implementation. 

- https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/obex-api.txt

muka, 2017-10-11:
Hi @BlackGh0st 
your contribution is  very welcome! 

OBEX support would be a great step forward on the library support of bluez API.

Just ping me if you need support in integrating or to adapt the current API to make thing work faster

muka, 2017-10-13:
Merged, thank you!

Regarding testing, I think a way would be to work on both sides (client+server) but still an idea

Regards

closed 2017-10-13 by muka