muka/go-bluetooth #23: [question] Can the library read beacons?

mikegleasonjr, 2017-11-21:


muka, 2017-11-21:
Hi, can you give some details? 

I guess it should not be much difficult to integrate, contribution is welcome

mikegleasonjr, 2017-11-21:
Hi @muka!

I don't know much about Bluetooth yet. There are specs around regarding beacons (iBeacon, Eddystone, etc) using Bluetooth LE. They just transmit a signal at regular intervals.

The signal is picked up by another Bluetooth device. That "signal", I don't know if it is proprietary or it is just a standard signal, like you can read them when you scan devices around you. it would be just a matter of decoding the packets according to the standard.

I just need to read beacons, on the receiving side. I will read more on the topic and report back. If you have any quick pointers, I am listening :)

muka, 2017-11-21:
Hi 
I actually have on my desk a few estimote devices which should do exactly that but have not had time to try out. 

I need to check better, would you like to see here

https://github.com/muka/go-bluetooth/blob/c23d2e9c9964ab9372e613c4934e5d3fd0f9c968/examples/watch_changes.go#L98

The discovery allows you to get RSSI and other device details, maybe not exactly what you are looking for but can give some result (RSSI is higher when device is near)

mikegleasonjr, 2017-11-21:
I really just need to know if a device is nearby, RSSI is good if I can estimate the distance also.

I can hardcode also my device IDs. I will also test on my end, I have an anndroid app that can broadcast beacons.

Let's see!

muka, 2018-01-13:
Closing, please reopen in case of need.

closed 2018-01-13 by muka