muka/go-bluetooth #24: Examples

estutzenberger, 2018-01-08:
Hi,

I'm trying to get started with the examples but have run into an issue. The API relies on tj/go-debug which is either a private repo or no longer exists. Did it move somewhere else? Is there a way to work around this issue?

Thanks,

Eric

muka, 2018-01-09:
Hi @estutzenberger the dependency disappeared a few day ago and had no time to upgrade the code yet. 

Last commits removed it at all (was nice for dev but not really useful to the project itself)

Let me know how it goes now 

Thanks
Luca

estutzenberger, 2018-01-09:
Awesome, thanks for the update. Looks like there are still some references to that package still hanging around in bluez/Client.go and emitter/events.go:

```# github.com/muka/go-bluetooth/bluez
../github.com/muka/go-bluetooth/bluez/Client.go:10:2: undefined: dbg
../github.com/muka/go-bluetooth/bluez/Client.go:33:3: undefined: dbg
../github.com/muka/go-bluetooth/bluez/Client.go:45:2: undefined: dbg
../github.com/muka/go-bluetooth/bluez/Client.go:63:2: undefined: dbg
../github.com/muka/go-bluetooth/bluez/Client.go:100:2: undefined: dbg
../github.com/muka/go-bluetooth/bluez/Client.go:126:2: undefined: dbg
../github.com/muka/go-bluetooth/bluez/Client.go:144:2: undefined: dbg
# github.com/muka/go-bluetooth/emitter
../github.com/muka/go-bluetooth/emitter/events.go:99:2: undefined: dbg
../github.com/muka/go-bluetooth/emitter/events.go:104:2: undefined: dbg
../github.com/muka/go-bluetooth/emitter/events.go:107:2: undefined: dbg
../github.com/muka/go-bluetooth/emitter/events.go:136:2: undefined: dbg
../github.com/muka/go-bluetooth/emitter/events.go:156:5: undefined: dbg
```

muka, 2018-01-09:
Oh sorry, was relying on the IDE feedback and  did not test properly. Should be all clean now

muka, 2018-01-09:
I have updated the code further, should be fine now

estutzenberger, 2018-01-09:
Great! I can build and test the examples now. Thanks!

closed 2018-01-09 by estutzenberger