muka/go-bluetooth #28: Nil pointer dereference on dbus channel

jaredwolff, 2018-01-22:
Hey @muka 

Have you seen this runtime error before? I seem to get it from time to time when initializing multiple devices. 

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x8 pc=0x19f600]

goroutine 1 [running]:
gatt/vendor/github.com/muka/go-bluetooth/bluez/profile.(*GattCharacteristic1).Register(0x0, 0x11, 0x11cb3d20, 0x1)
	/Users/jaredwolff/go/src/gatt/vendor/github.com/muka/go-bluetooth/bluez/profile/GattCharacteristic1.go:81 +0x14
main.(*Thingy).notify(0x1184e570, 0x11d0e900, 0x24, 0x11c94780, 0x1, 0x11d0e900)
```

muka, 2018-01-23:
Hi, not sure actually need to give a try. A patch is welcome if you spot the bug yourself. Thanks!

jaredwolff, 2018-01-23:
I'm getting some more similar errors 

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x19f70c]

goroutine 1 [running]:
gatt/vendor/github.com/muka/go-bluetooth/bluez/profile.(*GattCharacteristic1).ReadValue(0x0, 0x10dc2fe0, 0x0, 0x10dc2fe0, 0x0, 0x10dc2fe0, 0x24)
	/Users/jaredwolff/go/src/gatt/vendor/github.com/muka/go-bluetooth/bluez/profile/GattCharacteristic1.go:111 +0x60
main.SetBluetoothIntervals(0x10a0e5a0, 0x470c7b, 0x4)
...
```

They all seem to be surrounding bluez.Client. As if something inside is being changed or maybe the client disconnects at that very moment leaving nothing but some nil/dereferenced pointers. I'll play around with it more tomorrow.

muka, 2018-01-23:
No rampant errors! Totally agree 😁

jaredwolff, 2018-01-23:
Was there a reason why you didn't want to check if `deviceFound` in the GetCharByUUID function in Device.go?

I'm seeing some weirdness on a fresh boot that the devices can be found but when I go to subscribe to the UUIDs the chars don't exist

I added this in that function to at least catch the error:

```
	if deviceFound == nil {
    return nil, errors.New("Characteristic not found.")
	}
```

jaredwolff, 2018-01-23:
Also, the cache portion in the `GetCharsList()` of Device.go may be troublesome if we get into this state. As long as there is 1 UUID in the char list it will bypass finding new ones.

```
//GetCharsList return a device characteristics
func (d *Device) GetCharsList() []dbus.ObjectPath {

	var chars []dbus.ObjectPath

	if len(d.chars) != 0 {

		for objpath := range d.chars {
			chars = append(chars, objpath)
		}

		return chars
	}
```

muka, 2018-01-24:
Thank you maybe caching can be disabled until we reach a more stable setup. 

I will do some more test in the weekend

muka, 2018-01-25:
Hi @jaredwolff I've refactored a bit the api to add error handling as suggested. There may be some api breakage, but for good.  Let me know if you have any feedback

ps Really enjoying your blog. We do similar activities, would be nice to get in touch. Best!

closed 2018-01-25 by muka

jaredwolff, 2018-01-25:
@muka awesome. I looked it over quickly. Definitely a good call to catch these errors where they happen. I'll get it merged in with what I'm working on to see if anything breaks. 

Thanks for the kind words. It's been a labor of love for when I have an extra moment to nerd out. I think going into further detail on how to connect to BTLE devices via this library would be useful for the community and to gather more support around this project.