muka/go-bluetooth #34: interface conversion issue with the emitter

saurabh-newera, 2018-03-29:
while fetching sensor data:-
panic: interface conversion: interface {} is devices.SensorTagDataEvent, not api.DataEvent

goroutine 31 [running]:
main.ConnectAndFetchSensorDetailAndData.func1(0x75a180, 0xc4201e96c0)
	/home/saurabh/examples/sensortag/adpater.go:147 +0x1f7
created by github.com/muka/go-bluetooth/emitter.loop
	/home/saurabh/.go/src/github.com/muka/go-bluetooth/emitter/events.go:59 +0x205
exit status 2

saurabh-newera, 2018-03-29:
Issue fixed:
inside :- examples/sensortag_info/main.go
- x := ev.GetData().(api.DataEvent)
+ x := ev.GetData().(devices.SensorTagDataEvent)

structure definition has changed in backend, but not updated in example code.

saurabh-newera, 2018-03-29:
After 1-2 minutes sensors stop sending data(in connected state).
Sensor connectivity is also not stable. (After few minutes sensors got disconnected)

muka, 2018-04-18:
Thank you, fixed. The connectivity issues may be related to many factors and probably will require more research on your side.

commit e8afdeea (2018-04-18): review casting, fixes #34
commit 8b6a2b21 (2018-04-19): review casting, fixes #34
closed 2018-04-18 by muka