muka/go-bluetooth #42: Reading GATT Characteristic data from dbus

don, 2018-05-10:
I'm running the [service example](https://github.com/muka/go-bluetooth/tree/master/examples/service) on Ubuntu 17.10. The Bluetooth part works great. When I write data from my phone, I see it in the dbus-monitor (sudo dbus-monitor --system).

		method call time=1525902804.851076 sender=:1.2 -> destination=:1.90 serial=117 path=/org/bluez/example/service/service1/char1; interface=org.bluez.GattCharacteristic1; member=WriteValue
		   array of bytes "hello"
		   array [
		      dict entry(
		         string "device"
		         variant             object path "/org/bluez/hci0/dev_61_E0_93_AC_98_E7"
		      )
		   ]
		signal time=1525902804.852227 sender=:1.90 -> destination=(null destination) serial=18 path=/org/bluez/example/service/service1/char1; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
		   string "org.bluez.GattCharacteristic1"
		   array [
		      dict entry(
		         string "Value"
		         variant             array of bytes "hello"
		      )
		   ]
		   array [
		   ]
		
The problem is I don't know how to read the PropertiesChanged notification for /org/bluez/example/service/service1/char1 from DBus.

I created a similar example with [python-bluezero](https://github.com/ukBaz/python-bluezero). Using [pydbus](https://github.com/LEW21/pydbus), I'm able to read the Bluetooth data when my phone updates a characteristic.

        # python
		from pydbus import SystemBus, Variant
		from gi.repository import GLib

		bus = SystemBus()
		proxy = bus.get('ukBaz.bluezero', '/ukBaz/bluezero/service0001/char0001')
		proxy.onPropertiesChanged = print
		GLib.MainLoop().run()
		
I'm trying to use the same Python client to read data from the go-bluetooth sample but I don't know which bus name. org.bluez and go.bluetooth don't work.

		proxy = bus.get(BUS_NAME, '/org/bluez/example/service/service1/char1')

Any suggestions on which bus name to use?

muka, 2018-09-12:
Sorry for the long delay, the one you used seems fine. I would suggest to start looking at dbus permissions first and eventually at the module implementations to be sure there are no bug o side cases.

muka, 2019-02-05:
Closing, reopen if needed

closed 2019-02-05 by muka