muka/go-bluetooth #44: arch linux permission

vendelin8, 2018-06-08:
Hi,

I tried the example named discovery on Arch Linux x64, latest go 1.10.2-2, with turned on and enabled bluetooth, with sudo. At first, it failed at
```
a := linux.NewBtMgmt(adapterID)
err = a.Reset()
```

with
```
DEBU[0000] Exec: [btmgmt --index hci0 power off]        
ERRO[0000] exit status 1
```

I'm pretty sure that it was because of a permission problem, because I tried the same thing in a console with sudo, and it works without an error. But I though it's not that much needed, so I commented it out, and tried again. It has read the cached devices correctly, but then it failed with:
```
INFO[0000] Discovered devices:                          
ERRO[0000] Resource Not Ready
```

I think that it's a permission error too. What do you think about this?

denysvitali, 2018-08-16:
Have you followed the [Development Notes](https://github.com/muka/go-bluetooth#development-notes)?

vendelin8, 2018-08-16:
Yes, I tried it. The main problem is that hciconfig and such are deprecated. I installed it anyway from AUR, but then I still get errors, eg.:
$ bluetoothd
bash: bluetoothd: command not found

I found it here, that wasn't in the path:
/usr/lib/bluetooth/bluetoothd
so now it works, but it should be updated to current bluetooth commands.

closed 2018-08-16 by vendelin8