muka/go-bluetooth #48: Flutter mobile app

gedw99, 2018-09-10:
Flutter and golang can communicate over grpc well now.

So was thinking of making a flutter app for this.

What do people think ?

Flutter works on desktop and mobile.
It would talk to a raspberry pi in your house over grpc localhost.
That's my plan anyway.

I guess naming your Bluetooth devices and saving the mapping of the device name to the device Bluetooth ID would be an obvious need.

Pairing can be done on the raspberry pi and shown in your mobile etc. So you don't have to pair your phones with all your devices.

Also can use autocert and a tunnel so you can access everything remotely.

If anyone has other needs chime in !

muka, 2018-09-10:
Looks like a great idea, do you plan to open source it?

gedw99, 2018-09-10:
Yes of course.
Hoping to get a grpc API going with your advise first.  That's the first
step I feel. I thinking that a generic API could be also smart as there are
Bluetooth bulbs etc and the close software is scarey to me because so many
companies only eep the mobile software up to date for a while and then let
it die and slow death.

I was reading through the code and issues.
It looks like you are getting close to having client server architecture ?

gedw99, 2018-09-12:
> Yes it is there indeed, check the service example
> https://github.com/muka/go-bluetooth/tree/master/examples/service
>
> If you find out any improvement you are welcome

I now have webrtc data channels working between golang server and mobiles.

This will make it easier to reach your raspberry pi from outside your LAN
without exposing a public static IP on your raspberry pi.
Webrtc data channels can use protobuf and so I think I can reuse the
protobuf over webrtc.

The webrtc code is 100% golang based too except for the bi ding to openssl
for the DTLS. So it should be easy to build and deploy.

Feel free to chime in if you have any suggestions.

muka, 2018-10-12:
Closing as not an issue, please reopen it if needed. Thanks

closed 2018-10-12 by muka