muka/go-bluetooth #49: Adapter1 AdressType field for bluez 5.49

mame82, 2018-09-15:
Hi,

I'm currently using your code as reference for my own project.
First things first, this repo is great for everybody who's trying to find a simple way to interface with Bluez DBus API from native Go code (I'm currently wrapping Bluez-tools binaries to achieve my goals and this repo is a awesome guide to get rid of this).

Doing some tests I recognized that the `GetProperties` method of Adapter1 errors on Bluez 5.49.

This is because the `AddressType` field is missing in the respective struct and reflective casting of the map fails.

Could be that this field isn't present in older Bluez versions you support, but it seems you have to distinguish between Bluez versions to get casting to internal structs future proof.

Thanks for this nice project!

Cheers Mame82

muka, 2018-09-16:
Hi and thank you for the kind words :+1: 

As you told there is no real backward compatibility between APIs and  a I would not actually address it.

Maybe a simple way could be to just to tag the project and refer to bluez API compatibility in README.

If you want to to propose a PR to fix the error with the newest version I would be happy to review & merge

Thanks!

mame82, 2018-09-16:
The logic you're using here
https://github.com/muka/go-bluetooth/blob/master/util/map_struct.go#L32
iterates over the map provided by the DBus interface. In fact it would ignore every field of the target Go struct, which doesn't exist in the DBus properties.
If `AddressType string` would be added here
https://github.com/muka/go-bluetooth/blob/master/bluez/profile
/Adapter1.go#L33
it would simply be ignored on old Bluez version, but would get assigned in 5.49.

If a property will be removed in a newer bluez version, the behavior would basically be the same (target field won't be filled with data).

Considering a PR I don't want to make a quick shot. Haven't dived through all the code, but as stated I'm working on a custom implementation suiting my needs. If I'm able to come up with a better solution for "DBus interface to Go struct" mapping, I gonna file a PR.

niziak, 2018-10-12:
Fixed in 74174a9889d6bbb24b8e4ec4988b1be1b3ab7d67

muka, 2018-10-12:
Closing thanks!

closed 2018-10-12 by muka

mame82, 2018-10-12:
Thx for fix and sorry for the missing reply.

Have written my own bindings, meanwhile (less features, but support for bt-mgmt API)

https://github.com/mame82/mblue-toolz

Cheers