muka/go-bluetooth #51: Data race (api.(*Device).watchProperties / reflect.Value.Set())

niziak, 2018-10-03:
Hello.
This is my first approach to write something in golang, so maybe I'm using your library in wrong way.

I've enabled filtered scanning for advertisement with duplicated advertisement enabled, so I'm receiving each advertisement. 
Bluez signals changes on device for every advertisement. And the change set contains **two changes** at time (`RSSI` and `ManufacturerData` fields).
In this specified case (two changes at once), `for` loop in `api.(*Device).watchProperties` is processing changes one by one and emits `PropertyChangedEvent`. 

Problem is that this emited event from first change is concurrently processed and my callback is called, when  `api.(*Device).watchProperties` is still in `for` loop.
In my callback I want to read some other device properties, this is a moment where golang race detector complains (because `api.(*Device).watchProperties` is modifying device properties and in another routine my callback also want to do something with device properties)

niziak, 2019-02-05:
Please check my latest changes which works for me:
https://github.com/niziak/go-bluetooth/commits/niziak/devel
Almost all runtime issues was solved, but whole code should be reviewed again, because I'm not using all interfaces of this library. Also I'm a newbie in Golang, so maybe some my changes can look like reinventing the wheel.

In general: all data exposed to user are not a copy of data but pointers to internal data which can be changed in meantime (i.e. by object manager).

muka, 2019-02-05:
Would you pack the changes in a PR so I can merge it to the repo? 
If you consider it useful we can try to review the code and find improvements (if any would be needed)

Thank you

muka, 2019-02-05:
I added sync support based on your code, thank you! Closing for now, please reopen if needed

closed 2019-02-05 by muka