muka/go-bluetooth #56: ToMap() function in go-bluetooth/bluez/profile/GattCharacteristic1.go returns "GattCharacteristic1Properties: Descriptors contains an ObjectPath that is not valid" error for valid object path.

KSBhate, 2019-02-05:


muka, 2019-02-05:
Hi, can you provide a bit more context about what is happening? Some code to test would also help.

KSBhate, 2019-02-08:
In the function **ToMap** in **GattCharacteristic1.go**,   we check for the valid service path first :

`if !d.Service.IsValid() {
         return nil, errors.New("GattCharacteristic1Properties: Service ObjectPath is not valid")
}`

which is correct. We use _!d.Service.IsValid()_. i.e  (not d.Service.IsValid()).

But while checking for the valid descriptors, I think, **!** (not ) is missing ahead of _if d.Descriptors[i].IsValid()_. which makes a valid descriptor throw an error and we end up getting an empty map.

commit e0740c96 (2019-02-15): fix check on valid descriptor, @KSBhate fixes #56
closed 2019-02-15 by muka