muka/go-bluetooth #60: Advertising multiple services failed

KSBhate, 2019-02-08:
We are using the master branch from the go-bluetooth.
We have added the services as per the examples.
When we have only one service the code works fine and we can see our service in the ble-client app.
But as soon as we add another service in the same way and then we start advertising, we get error while registering the advertise.: RegisterAdvertisement, and application doesn't advertise anything.

In the module Application.go, inside function StartAdvertising -> RegisterAdvertisement, where we ultimately give a dbus call for register adv, which is failing.
Exact error message: Failed to parse advertisement
Can someone point out what is wrong or why would it happen?
It will be a great help.

muka, 2019-02-15:
I will check, if you can post some testable code would help me a lot!

KSBhate, 2019-02-16:
This is not an issue with go-bluetooth library. I was making a mistake while creating GATT service.

While advertising service, I got an error in bluetooth service - "Advertising data too long or couldn't be generated." I read that advertising payload is 0-31 bytes long. I was trying to advertise multiple custom services having 128-bit UUID. These UUIDs were getting added in advertisement data and advertisement data length was exceeding maximum length.

Problem is solved by setting parameter "advertisedOptional" as "false" while creating service. 

	`customService, err := app.CreateService(serviceProperties, false)		`

commit d8729b50 (2019-02-17): added warning when adv data is over 31 bytes, close #60
closed 2019-02-17 by muka