muka/go-bluetooth #63: Agent example returns empty device list all the time

vladyslav2, 2019-02-15:
Just trying to run a simple example with an agent and I notice
```
	devices, err := api.GetDevices()
	if err != nil {
		log.Error("GetDevices", err)
		return
	}
	log.Info("devices", devices)
```

`api.GetDevices()` will always return an empty list
Any ideas why/what I might doing wrong?

and follow up question 
```
	AGENT_PATH      = "/gameshell/bleagentgo"
```

where can I get `bleagento`? do I need to compile it myself and put into /gameshell/ folder?

muka, 2019-02-15:
Hi
that "path" is a dbus path and is registered in the code see

https://github.com/muka/go-bluetooth/blob/master/examples/agent/main.go#L105

If you get an empty list of devices chances are that the BLE discovery has not run yet.

You could try running the discovery example to see if something is found around or use another tool for starting the discovery

vladyslav2, 2019-02-15:
@muka  thank you for a quick response and a this library
it worked

muka, 2019-02-18:
That's great, need to provide better docs or improved API. Closing, please reopen if needed.

closed 2019-02-18 by muka