muka/go-bluetooth #65: How to properly reconnect to the device?

jsnjack, 2019-02-18:
Hi,

First of all, many thanks for the library! I have notice that one of my bluetooth devices may disconnect once in a while. Sometimes I'm able to reconnect to the device and when I do, I start to receive double notifications for all events. 

What would be the correct way to reconnect to the device?

muka, 2019-02-19:
Hi, this is indeed a bug. The event is emitted twice as the reconnect does create a doubled subscription.

Will check it

muka, 2019-03-24:
Sorry, time run fast and I had not yet time to fix.

Have you found a fix you could commit as a PR eventually?

jsnjack, 2019-03-28:
No, unfortunately, not yet.. Haven't been able to get back to the project for some time :(

muka, 2019-05-11:
Hi
i tried to check over this but cannot replicate.

`On('changed')` unregister then register again to event notifications so duplication should not happens

https://github.com/muka/go-bluetooth/blob/master/api/Device.go#L279

Can provide some test code so I can check ?

Closing for now, reopen when needed. Thanks

closed 2019-05-11 by muka