muka/go-bluetooth #66: Any way I can connect to the device without waiting for discover service?

vladyslav2, 2019-02-21:
If I know device-id do I always have to wait while discover service will find that device?
I.E. do I need to do all the time:
```
        devices, err := api.GetDevices()
	if err != nil {
		log.Errorf("GetDevices: %s", err)
		return
	}

	log.Info(devices)

	for i, v := range devices {
		path := strings.Replace(devMac, ":", "_", -1)
		if strings.Contains(v.Path, path) {
			log.Info(i, v.Path)
			log.Info("Pairing...")
			err := v.Pair()
			if err == nil {
				log.Info("Pair succeed,Connecting...")
				setTrusted(dev_mac_path)
				v.Connect()
			} else {
				log.Error("Pair failed:", err)
			}
		}
	}
```

Or there are a way to connect  to device without waiting while device will show up at `api.GetDevices()` ?

lets say:
```
dev := api.NewDevice(path)
dev.Connect()
```
?

vladyslav2, 2019-02-21:
and follow up question If I will start application, connect to device. Next time I start application - do I have to put my device in the pair mode all the time? Is there a way to remember device between application restart?

vladyslav2, 2019-02-21:
hm... seems like linux bluetooth library doing that automatically for me if I will markdown devices as trusted

muka, 2019-02-23:
Once trusted device are cached, as long you do not remove them via API (in the demo code this is done on purpose to simplify the development)

muka, 2019-03-24:
I am closing due to no activity, please reopen of needed. Thanks

closed 2019-03-24 by muka