muka/go-bluetooth #67: watch-agent for heart rate device

vladyslav2, 2019-02-22:
Hi
@muka  any advice what should I do in order to read data from heart rate device?

I'm using your watch-agent example with few modifications

Problem:
- [ ] I do have a lot of services and I don't know how to read data from it
- [ ] I'm getting bunch of  errors  No such interface 'org.bluez.GattCharacteristic1' for ` profile.NewGattCharacteristic1(charEvent.DevicePath)`

and currently i'm out of ideas how to fix them
 
```
$ go run cmd/console/main.go 
{"level":"debug","service":"heartrate","component":"discovery","time":"2019-02-22T13:14:04-08:00","message":"Reset bluetooth device"}

{"level":"info","service":"heartrate","component":"discovery","time":"2019-02-22T13:14:12-08:00","message":"name=TICKR 3988 addr=E7:BC:BE:49:92:0C rssi=-94"}
Found TICKR 3988 [addr:E7:BC:BE:49:92:0C], list profiles
Connecting device TICKR 3988

char data, props {/org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0008/char0009 /org/bluez/hci0/dev_E7_BC_BE_49_92_0C 0xc4200dd3b0 0} &{[] false false false /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0008 00002a05-0000-1000-8000-00805f9b34fb [indicate] []}
service  2A05 
Found char  (2A05 : /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0008/char0009)
char data, props {/org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0010/char0011 /org/bluez/hci0/dev_E7_BC_BE_49_92_0C 0xc4202ae380 0} &{[] false false false /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0010 00002a29-0000-1000-8000-00805f9b34fb [read] []}
service  2A29 
Found char  (2A29 : /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0010/char0011)
char data, props {/org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0010/char0013 /org/bluez/hci0/dev_E7_BC_BE_49_92_0C 0xc4202ae690 0} &{[] false false false /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0010 00002a27-0000-1000-8000-00805f9b34fb [read] []}
service  2A27 
Found char  (2A27 : /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0010/char0013)
char data, props {/org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0010/char0015 /org/bluez/hci0/dev_E7_BC_BE_49_92_0C 0xc4202d8690 0} &{[] false false false /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0010 00002a26-0000-1000-8000-00805f9b34fb [read] []}
service  2A26 
Found char  (2A26 : /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0010/char0015)
char data, props {/org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0017/char0018 /org/bluez/hci0/dev_E7_BC_BE_49_92_0C 0xc4202d89a0 0} &{[] false false false /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0017 a026e002-0a7d-4ab3-97fa-f1500f9feb8b [write-without-response notify] []}
service  E002 
Found char  (E002 : /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0017/char0018)
char data, props {/org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0017/char001b /org/bluez/hci0/dev_E7_BC_BE_49_92_0C 0xc420189490 0} &{[] false false false /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0017 a026e004-0a7d-4ab3-97fa-f1500f9feb8b [notify] []}
service  E004 
Found char  (E004 : /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0017/char001b)
char data, props {/org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service001e/char001f /org/bluez/hci0/dev_E7_BC_BE_49_92_0C 0xc4200ddce0 0} &{[] false false false /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service001e a026e00a-0a7d-4ab3-97fa-f1500f9feb8b [write-without-response notify] []}
service  E00A 
Found char  (E00A : /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service001e/char001f)
char data, props {/org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0022/char0023 /org/bluez/hci0/dev_E7_BC_BE_49_92_0C 0xc420189ab0 0} &{[] false false false /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0022 00002a37-0000-1000-8000-00805f9b34fb [notify] []}
service  2A37 
Found char  (2A37 : /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0022/char0023)
char data, props {/org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0022/char0026 /org/bluez/hci0/dev_E7_BC_BE_49_92_0C 0xc420366000 0} &{[] false false false /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0022 00002a38-0000-1000-8000-00805f9b34fb [read] []}
service  2A38 
Found char  (2A38 : /org/bluez/hci0/dev_E7_BC_BE_49_92_0C/service0022/char0026)
err:  No such interface 'org.bluez.GattCharacteristic1'
err:  No such interface 'org.bluez.GattCharacteristic1'
err:  No such interface 'org.bluez.GattCharacteristic1'
err:  No such interface 'org.bluez.GattCharacteristic1'
err:  No such interface 'org.bluez.GattCharacteristic1'
err:  No such interface 'org.bluez.GattCharacteristic1'
err:  No such interface 'org.bluez.GattCharacteristic1'
err:  No such interface 'org.bluez.GattCharacteristic1'
err:  No such interface 'org.bluez.GattCharacteristic1'
```

If you have any suggestions - that will be nice

thank you

muka, 2019-02-23:
Hi
I can suggest, if you do not have it already, to install `d-feet` which can help you in inspecting the DBus interfaces.

Try looking at the SensorTag implementation 
https://github.com/muka/go-bluetooth/tree/master/devices

It read, write and subscribe to device services/characteristic it should provde you some guidance

Reading data may not be obvious, if your devices have a custom way to marshal data you need to know how to decode (or reverse engineer) the payload encoding.

jsnjack, 2019-02-23:
You can find description for most of characteristics here https://www.bluetooth.com/specifications/gatt/ . The one that you are interested in is https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.heart_rate_measurement.xml

If you are using Linux, you can also connect and explore the device with bluetoothctl

muka, 2019-03-24:
Thanks for the hint  @jsnjack, it should live in the readme also.

I am closing this issue for no activity, please reopen if needed.

Thanks

closed 2019-03-24 by muka