muka/go-bluetooth #69: RegisterAdvertisement dbus errors on bluez 5.50

davidknezic, 2019-03-24:
I've managed to integrate `go-bluetooth` and run a GATT service on my Raspberry Pi 3 with Raspbian. (`bluez 5.43`)

Now I'm trying to get it to work on a yocto based image. (`bluez 5.50`)

However, during the call to `RegisterAdvertisement` some things don't seem to work properly, looking at the dbus logs:

```
error time=1552854045.795385 sender=org.freedesktop.DBus -> destination=:1.7 
error_name=org.freedesktop.DBus.Error.ServiceUnknown reply_serial=216
   string "The name :1.15 was not provided by any .service files"
method call time=1552854045.795651 sender=:1.7 -> destination=:1.15 serial=217 path=/org/bluez/advertisement/hci0; interface=org.freedesktop.DBus.ObjectManager; member=GetManagedObjects
error time=1552854045.795783 sender=:1.15 -> destination=:1.7 error_name=org.freedesktop.DBus.Error.UnknownInterface reply_serial=217
   string "Object does not implement the interface"
```

<details><summary>Full dbus log excerpt </summary><code><pre>
method call time=1552854045.763240 sender=:1.15 -> destination=org.bluez serial=11 path=/org/bluez/hci0; interface=org.bluez.LEAdvertisingManager1; member=RegisterAdvertisement
   object path "/org/bluez/advertisement/hci0"
   array [
   ]
method call time=1552854045.767510 sender=:1.7 -> destination=org.freedesktop.DBus serial=211 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=GetNameOwner
   string ":1.15"
method return time=1552854045.767784 sender=org.freedesktop.DBus -> destination=:1.7 serial=132 reply_serial=211
   string ":1.15"
method call time=1552854045.767969 sender=:1.7 -> destination=org.freedesktop.DBus serial=212 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=AddMatch
   string "type='signal',sender=':1.15',path='/org/bluez/advertisement/hci0',interface='org.freedesktop.DBus.ObjectManager',member='InterfacesAdded'"
method return time=1552854045.768161 sender=org.freedesktop.DBus -> destination=:1.7 serial=133 reply_serial=212
method call time=1552854045.771002 sender=:1.7 -> destination=org.freedesktop.DBus serial=213 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=AddMatch
   string "type='signal',sender=':1.15',path='/org/bluez/advertisement/hci0',interface='org.freedesktop.DBus.ObjectManager',member='InterfacesRemoved'"
method return time=1552854045.771412 sender=org.freedesktop.DBus -> destination=:1.7 serial=134 reply_serial=213
method call time=1552854045.779049 sender=:1.7 -> destination=org.freedesktop.DBus serial=214 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=AddMatch
   string "type='signal',sender=':1.15',path_namespace='/org/bluez/advertisement/hci0'"
method return time=1552854045.779367 sender=org.freedesktop.DBus -> destination=:1.7 serial=135 reply_serial=214
method call time=1552854045.779492 sender=:1.7 -> destination=org.freedesktop.DBus serial=215 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=AddMatch
   string "type='signal',sender=':1.15',path='/org/bluez/advertisement/hci0',interface='org.freedesktop.DBus.Properties',member='PropertiesChanged',arg0='org.bluez.LEAdvertisement1'"
method return time=1552854045.779680 sender=org.freedesktop.DBus -> destination=:1.7 serial=136 reply_serial=215
method call time=1552854045.793739 sender=:1.7 -> destination=org.freedesktop.DBus serial=216 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=StartServiceByName
   string ":1.15"
   uint32 0
error time=1552854045.795385 sender=org.freedesktop.DBus -> destination=:1.7 error_name=org.freedesktop.DBus.Error.ServiceUnknown reply_serial=216
   string "The name :1.15 was not provided by any .service files"
method call time=1552854045.795651 sender=:1.7 -> destination=:1.15 serial=217 path=/org/bluez/advertisement/hci0; interface=org.freedesktop.DBus.ObjectManager; member=GetManagedObjects
error time=1552854045.795783 sender=:1.15 -> destination=:1.7 error_name=org.freedesktop.DBus.Error.UnknownInterface reply_serial=217
   string "Object does not implement the interface"
method call time=1552854045.795967 sender=:1.7 -> destination=:1.15 serial=218 path=/org/bluez/advertisement/hci0; interface=org.freedesktop.DBus.Properties; member=GetAll
   string "org.bluez.LEAdvertisement1"
</pre></code>
</details>

This is the code that triggers these errors: [the-lightning-land/sweetd/bluetooth/pairing/controller_linux.go#L156](https://github.com/the-lightning-land/sweetd/blob/bluetooth/pairing/controller_linux.go#L156)

Is it possible that something has changed in `bluez` between `5.43` and `5.50` that leads to the `Object does not implement the interface` error? Or could there be something else that might lead to this problem?

Help very appreciated!

muka, 2019-03-24:
Hi David
have you added the dbus permissions definition files ?

Samples are here https://github.com/muka/go-bluetooth/tree/master/scripts

If not the case, can you try to have a setup with 5.50 and run a plain example to see if this is still happening?

Let me know so we can investigate further

Thanks

davidknezic, 2019-03-24:
Hi @muka, thanks for helping out!

Yes, I have removed the standard bluetooth permissions file and replaced it with the two that are in the `scripts` folder (with some additions):

<details>
<summary><code>/etc/dbus-1/system.d/dbus-go-bluetooth-service.conf</code></summary>

```
<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN" "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>
  <policy user="root">
    <allow own="go.bluetooth"/>
    <allow send_destination="go.bluetooth"/>
    <allow send_destination="org.bluez"/>
  </policy>
  <policy at_console="true">
    <allow own="go.bluetooth"/>
    <allow send_destination="go.bluetooth"/>
    <allow send_destination="org.bluez"/>
  </policy>
  <policy context="default">
    <deny send_destination="go.bluetooth"/>
  </policy>
</busconfig>
```

</details>

<details>
<summary><code>/etc/dbus-1/system.d/dbus-go-bluetooth-service.conf</code></summary>

```
<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN" "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">

<busconfig>

    <policy user="root">

        <allow own="go.bluetooth"/>
        <allow own_prefix="go.bluetooth"/>
        <allow send_destination="go.bluetooth" send_interface="go.bluetooth"/>
        <allow send_destination="go.bluetooth" send_interface="org.freedesktop.DBus.Introspectable"/>
        <allow send_destination="go.bluetooth" send_interface="org.freedesktop.DBus.Properties"/>
        <allow send_destination="go.bluetooth" send_interface="org.freedesktop.DBus.ObjectManager"/>
        <allow send_destination="go.bluetooth"/>

        <allow own="org.bluez"/>
        <allow own_prefix="org.bluez"/>
        <allow send_destination="org.bluez" send_interface="org.bluez"/>
        <allow send_destination="org.bluez" send_interface="org.freedesktop.DBus.Introspectable"/>
        <allow send_destination="org.bluez" send_interface="org.freedesktop.DBus.Properties"/>
        <allow send_destination="org.bluez" send_interface="org.freedesktop.DBus.ObjectManager"/>
        <allow send_destination="org.bluez"/>

        <allow send_interface="org.freedesktop.DBus.ObjectManager"/>
        <allow send_interface="org.freedesktop.DBus.Properties"/>
        <allow send_interface="org.freedesktop.DBus.Introspectable"/>
        <allow send_interface="org.bluez.GattCharacteristic1"/>
        <allow send_interface="org.bluez.GattDescriptor1"/>
        <allow send_interface="org.bluez.GattService1"/>
        <allow send_interface="org.bluez.GattManager1"/>
        <allow send_interface="org.bluez.LEAdvertisement1"/>
        <allow send_interface="org.bluez.Agent1"/>
        <allow send_interface="org.bluez.MediaEndpoint1"/>
        <allow send_interface="org.bluez.MediaPlayer1"/>
        <allow send_interface="org.bluez.Profile1"/>
        <allow send_interface="org.bluez.LEAdvertisingManager1"/>
    </policy>
    <policy at_console="true">
     <allow send_destination="org.bluez"/>
    </policy>

    <policy context="default">
    </policy>

</busconfig>
```

</details>

This was the original that was added by the yocto build. I've replaced it entirely because I got a couple of permission errors while calling `go-bluetooth`.

<details>
<summary><code>/etc/dbus-1/system.d/bluetooth.conf</code></summary>

```
<!-- This configuration file specifies the required security policies
     for Bluetooth core daemon to work. -->

<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>

  <!-- ../system.conf have denied everything, so we just punch some holes -->

  <policy user="root">
    <allow own="org.bluez"/>
    <allow send_destination="org.bluez"/>
    <allow send_interface="org.bluez.Agent1"/>
    <allow send_interface="org.bluez.MediaEndpoint1"/>
    <allow send_interface="org.bluez.MediaPlayer1"/>
    <allow send_interface="org.bluez.Profile1"/>
    <allow send_interface="org.bluez.GattCharacteristic1"/>
    <allow send_interface="org.bluez.GattDescriptor1"/>
    <allow send_interface="org.bluez.LEAdvertisement1"/>
    <allow send_interface="org.freedesktop.DBus.ObjectManager"/>
    <allow send_interface="org.freedesktop.DBus.Properties"/>
  </policy>

  <policy at_console="true">
    <allow send_destination="org.bluez"/>
  </policy>

  <!-- allow users of lp group (printing subsystem) to 
       communicate with bluetoothd -->
  <policy group="lp">
    <allow send_destination="org.bluez"/>
  </policy>

  <policy context="default">
    <deny send_destination="org.bluez"/>
  </policy>

</busconfig>
```

</details>

muka, 2019-03-26:
Can you have a try with an example from the repository? It could be easier for me to replicate.

davidknezic, 2019-03-26:
Yes, sure! Just tried the `service` example. It leads to the same error.

```
root@raspberrypi0-wifi:/data# ./service 
INFO[0000] Application started, waiting for connections 
DEBU[0000] Registering app /go_bluetooth/example/service 
DEBU[0000] Registering service on hci0 (/org/bluez/advertisement/hci0) 
WARN[0000] Advertisment limit of 31 bytes may have been exceeded. Consider not exposing all services IDs with `CreateService(props, false)` 
error time=1552893110.036474 sender=org.freedesktop.DBus -> destination=:1.0 error_name=org.freedesktop.DBus.Error.ServiceUnknown reply_serial=491
   string "The name :1.16 was not provided by any .service files"
error time=1552893110.044188 sender=:1.16 -> destination=:1.0 error_name=org.freedesktop.DBus.Error.UnknownInterface reply_serial=492
   string "Object does not implement the interface"
error time=1552893110.073839 sender=:1.16 -> destination=:1.0 error_name=org.freedesktop.DBus.Error.UnknownMethod reply_serial=494
   string "Unknown / invalid method"
error time=1552893110.074099 sender=org.freedesktop.DBus -> destination=:1.16 error_name=org.freedesktop.DBus.Error.AccessDenied reply_serial=17
   string "Rejected send message, 1 matched rules; type="error", sender=":1.16" (uid=0 pid=498 comm="./service ") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.UnknownMethod" requested_reply="0" destination=":1.0" (uid=0 pid=317 comm="/usr/libexec/bluetooth/bluetoothd ")"
INFO[0000] Application registered and advertising.
```

muka, 2019-03-27:
Seems an issue in the configuration of the system / yocto build. 

To recap, your user is `root` (matching `user=root` policy) and your exposed service map in dbus to `go.bluetooth`. 

In the code you could try to play with the object path / service name
https://github.com/the-lightning-land/sweetd/pull/14/files#diff-2650556a5ae8d1b4833240d0c10b6238R15

and ensure they match with the one in your config eg.
```
     <allow own="go.bluetooth"/> 
        <allow own_prefix="go.bluetooth"/>
```

Nice project the blockchain based candy dispenser, I want one of them! :+1:

davidknezic, 2019-03-27:
>In the code you could try to play with the object name / object path

That's a good suggestion, I will try to adjust the object name / object path and see if the policies match.

I find it very hard to find information regarding the dbus gatt API and how to use it to expose BLE services.

Is there anything I have to keep in mind when selecting `objectName` and `objectPath`? In the example the `objectName` is set to `org.bluez`, is that mandatory? Also the `objectPath` in the example is set to `/go_bluetooth/example/service`. Does that map to the `<allow own="go.bluetooth"/> ` in some way?

I really appreciate your help @muka 🙌 

>Nice project the blockchain based candy dispenser

Thank you!

>I want one of them! 👍

I think we can make that happen ☺️

davidknezic, 2019-03-29:
Hi @muka, just wanted to update you that I wasn't able to identify the problem so far. That's why I've tried downgrading the bluez5 package to 5.47. With that everything works very well.

I hope I'll have time again to investigate what changes between 5.47 and 5.50 there were that caused my problems, it won't be a huge priority though.

If someone stumbles over this issue and has had experience with `go-bluetooth` and bluez 5.50, it would be interesting to learn how it went!

muka, 2019-03-29:
I will check as on a plain Ubuntu seems to work. Can you share something I
can try, also on a py eventually

Thanks

sent from my phone

muka, 2019-05-11:
Hi, after upgrading to ubuntu 19.04 I was having the same issue on 5.50. 

To overcome I updated go deps (already in master) and added back the links for dbus configurations (as described in README)

I am closing the issue, please reopen if needed

Thank you

closed 2019-05-11 by muka