muka/go-bluetooth #73: Support battery level in bluez 5.48+

anonymouse64, 2019-04-23:
Starting in bluez 5.48, the battery level GATT is now exposed in dbus interface `org.bluez.Battery1` (see https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/battery-api.txt)

I can submit a PR implementing support for this (my initial idea was to add a `profile.Battery1` struct and `profile.Battery1Properties` and then put those inside the `api.Device` and expose it with `dev.GetBatteryProperties()` etc.) but the one thing I'm not sure about is how to handle bluez versions less than 5.48 properly. Is there currently anywhere in the code-base where the bluez version is checked or stored so I could check if that exists when initializing an `api.Device`? It's also not clear to me that we can easily detect the version of bluez via dbus alone unfortunately so the best thing I can guess is just silently continue on creating the device if the `Battery1` interface doesn't exist the first time we use that client.

muka, 2019-04-23:
Hi @anonymouse64 !
as of now there is no real support for old version of bluez. Having limited time to maintain the library, the attempt is to stay compatible in master with the most up to date (5.50 as of now).

Probably would be better to start tagging versions and try to match bluez version in a compatibility matrix. I will take a look to it.

If you provide a PR I will be glad to review and merge it. I would ask you to add also an example under `./examples` and update the README accordingly.

Thank you

anonymouse64, 2019-04-23:
Okay, I'll try to get that together this week sometime then.

muka, 2019-08-02:
Hi 
the interface is available here https://github.com/muka/go-bluetooth/tree/master/bluez/profile/battery

Please reopen if needed

closed 2019-08-02 by muka

muka, 2019-08-07:
You can find an usage example at 

https://github.com/muka/go-bluetooth/blob/master/examples/sensortag_info/sensortag_info.go#L68-L78