muka/go-bluetooth #74: Add support for digital and analog characteristics.

KSBhate, 2019-05-07:
Hi,

It seems that currently digital and analog characteristics for automation IO service are not supported in go-bluetooth library.

While using automation IO service, we can have multiple digital and analog characteristics. Each digital and analog characteristic has UUID 0x2A56 and 0x2A58 respectively.

Problem is that, currently it is not possible to identify that which analog or digital characteristic has been changed. This is because, we have application callbacks where we use characteristic UUID to identify the characteristic that has been changed.

//GattWriteCallback A callback we can register to handle write requests
type GattWriteCallback func(app *Application, service_uuid string, charUUID string, value []byte) error

//GattDescriptorWriteCallback A callback we can register to handle descriptor write requests
type GattDescriptorWriteCallback func(app *Application, service_uuid string, charUUID string, descUUID string, value []byte) error

//GattReadCallback A callback we can register to handle read requests
type GattReadCallback func(app *Application, service_uuid string, charUUID string) ([]byte, error)

//GattDescriptorReadCallback A callback we can register to handle descriptor ead requests
type GattDescriptorReadCallback func(app *Application, service_uuid string, charUUID string, descUUID string) ([]byte, error)

To fix this problem, I think, dbus.ObjectPath can be used instead of UUIDs of service, characteristic and descriptor. DBus object paths are guaranteed to be unique.

That is,

type GattWriteCallback func(app *Application, srviceObjPath dbus.ObjectPath, charObjPath dbus.ObjectPath, value []byte) error

type GattReadCallback func(app *Application, srviceObjPath dbus.ObjectPath, charObjPath dbus.ObjectPath) ([]byte, error)

muka, 2019-05-08:
Thank you for reporting, I agree on the point. Can you provide a PR to address this issue?

Thank you

KSBhate, 2019-05-09:
Hi,
Please refer PR https://github.com/muka/go-bluetooth/pull/75.

muka, 2019-05-11:
PR merged, thank you!

closed 2019-05-11 by muka