muka/go-bluetooth #76: "dbus: connection closed by user" when running discovery example

eeropenttinen, 2019-05-17:
First time when I run the example, it works fine and scans available Bluetooth devices.
Second run (after stopping the app with Ctrl C) will print out following:

> DEBU[0000] Reset bluetooth device
> ERRO[0000] FlushDevices.RemoveDevice dbus: connection closed by user

I added a graceful exit to the example to ensure api.Exit() was called but I got the same issue.

I'm running on following environment:
bluez version 5.48
Ubuntu 18.04.1 LTS
Go version 1.12.5 

After couple of minutes, the issue disappears for one run, but second run will cause the same issue.

eeropenttinen, 2019-07-19:
This issue is likely related to this: https://github.com/godbus/dbus/issues/144 - the error is the same.

muka, 2019-07-19:
Thanks
seeems we need to avoid the conn close until the program ends

I am going to check in the weekend

closed 2019-07-20 by muka

eeropenttinen, 2019-07-21:
Thanks. Based on changes in the commit, this will address lot of issues I had with the go-bluetooth library.

muka, 2019-07-21:
Thank you for finding that out! 

Please reopen if needed

commit 10238848 (2019-07-20): do not drop system bus connection, fixes #76