muka/go-bluetooth #81: Creating Eddystone Beacon

ukBaz, 2019-07-23:
Hello,

I am looking to create an Eddystone URL beacon using BlueZ DBus API. It looks like this library has some support for this. However, I am new to the Go language and am rather intimidated by the empty text editor in front of me. Is it possible for you to give me some guidance on getting started with an example for this?

An example of the information required for "org.bluez.LEAdvertisement1" 
Type = "broadcast"
ServiceUUIDs = "FEAA"
ServiceData = map[string[13]int] {"FEAA":  [16 0 1 98 108 117 101 116 111 111 116 104 7]}
IncludeTxPower = false

The DBus object path for the above then needs to be passed to method RegisterAdvertisement on interface "org.bluez.AdvertisementManager1"

Any pointers in getting started with this would be appreciated.

Thanks.

muka, 2019-07-31:
Hi @ukBaz  
I tried to get something to work and now I feel intimidate too :D
I am doing some refactoring on the  library, after that I will retry to setup a beaconing example.

I think the way you described is a pretty good start.

Let me know if you have any update with some code to work on. Once it work could you also submit a PR?

Thanks

muka, 2019-08-03:
Hello
I added an example here
https://github.com/muka/go-bluetooth/blob/master/examples/advertise/advertise.go

It shows on mobile app but I think it still need some refinement, as beacons are not constantly visible. 
Let me know

ukBaz, 2019-08-03:
Thanks for posting this. I did panic when I clicked on the link in the email and it said the link was broken but I see you have edited the location.  :-)

I am a golang noob so could you give a little more detail of how to run this. I have tried using your Makefile to run some other example but it is giving me various errors. Not sure if it is my setup or if there is something missing.

Thanks.

muka, 2019-08-04:
You can try with

`go run main.go beacon eddystone`

See examples folder for the source code

sent from my phone

ukBaz, 2019-08-04:
Thank you for sending the command across, very helpful.
This now suggests the beacon is running 
```
$ go run main.go beacon eddystone

DEBU[0000] Retrieving adapter instance hci0             
DEBU[0000] Connecting to DBus                           
DEBU[0000] Creating LEAdvertisement1 instance           
DEBU[0000] Exposing LEAdvertisement1 instance           
DEBU[0000] Setup adapter                                
DEBU[0000] Registering LEAdvertisement1 instance        
DEBU[0000] eddystone ready                  
```
When I use the nRF Connect app on my phone I can't see the beacon.
If I monitor with bluetoothctl then I see it registers the advertisement (ActivieInstances increases) but then removes it right away.
```
[CHG] Controller FC:F8:AE:8F:0C:A4 DiscoverableTimeout: 0x00000000
[CHG] Controller FC:F8:AE:8F:0C:A4 SupportedInstances: 0x04
[CHG] Controller FC:F8:AE:8F:0C:A4 ActiveInstances: 0x01
[CHG] Controller FC:F8:AE:8F:0C:A4 SupportedInstances: 0x05
[CHG] Controller FC:F8:AE:8F:0C:A4 ActiveInstances: 0x00
```

If I put `bluetoothd` in debug mode then I see that it is getting a client timeout
```
bluetoothd[2592]: src/adapter.c:set_discoverable() sending set mode command for index 0
bluetoothd[2592]: src/adapter.c:set_mode() sending set mode command for index 0
bluetoothd[2592]: src/advertising.c:register_advertisement() RegisterAdvertisement
bluetoothd[2592]: src/advertising.c:client_create() Adding proxy for /org/bluez/example/advertisement0
bluetoothd[2592]: src/advertising.c:register_advertisement() Registered advertisement at path /org/bluez/example/advertisement0
bluetoothd[2592]: src/advertising.c:parse_service_uuids() Adding ServiceUUID: FEAA
bluetoothd[2592]: src/advertising.c:parse_service_data() Adding ServiceData for FEAA
bluetoothd[2592]: src/advertising.c:parse_discoverable() Adding Flags 0x00
bluetoothd[2592]: src/advertising.c:refresh_adv() Refreshing advertisement: /org/bluez/example/advertisement0
bluetoothd[2592]: src/advertising.c:add_adv_callback() Advertisement registered: /org/bluez/example/advertisement0
bluetoothd[2592]: src/advertising.c:client_timeout() 
bluetoothd[2592]: src/advertising.c:client_release() Releasing advertisement :1.173, /org/bluez/example/advertisement0
```

I suspect that is `RegisterAdvertisement` getting a timeout trying to read the `org.bluez.LEAdvertisement1` interface. You say that beacons are not constantly visible, are you seeing the same?
Could it be that the publishing of `org.bluez.LEAdvertisement1` needs to happen in a separate thread? 

It is clear that my golang skills are not very good at the moment so please feel free to close this issue if you think the issue is with the user rather than the library.

muka, 2019-08-04:
There is still something out of place, the timeout is a good indicator.

This setup is similar to the bluez python example, it could be related to golang types introspection by dbus library.

Let's keep the issue open until it works properly, thanks

muka, 2019-08-07:
You were right, the timeout was the problem.  Removing `DiscoverableTimeout`  which was set to 0 show the beacon in a stable manner.

Maybe this is relevant https://www.spinics.net/lists/linux-bluetooth/msg79915.html

Please reopen if still not  working

Thanks

ukBaz, 2019-08-07:
Thanks for the update. I'll test in the next couple of days.

There is also this patch that I wondered if it was relevant:
https://marc.info/?l=linux-bluetooth&m=156509924619419&w=2

ukBaz, 2019-08-10:
Apologies, this took longer for me to get back to than I had hoped for.
Sadly, the beacon example still isn't working for me  :-(

[image]

I am busy on another project now so there is no pressure from me. However, if you would like me to test anything then let me know.

muka, 2019-08-11:
Thank you for the update, there is no hurry. I am working a bit on it as I
have some free time. If you have a change to try it up please let me know

sent from my phone

muka, 2019-08-11:
Just pushed the up date version which works with both ibeacon and eddystone.

https://github.com/muka/go-bluetooth/blob/master/examples/beacon/beacon.go#L14

Please reopen if needed

commit e70bdeb6 (2019-08-07): set beaconing timeout, fix #81
commit 0b96d6aa (2019-08-11): fix beaconing setup for both eddystone and ibeacon. Fix #81
closed 2019-08-11 by muka