muka/go-bluetooth #84: Issues w/ org.bluez.Profile1 and org.bluez.ProfileManager1 code

jaymell, 2019-08-19:
I've been trying to get code using the Profile1 and ProfileManager1 helper functions and not having much luck. Sample code:

```import (
  "github.com/muka/go-bluetooth/bluez/profile/profile"
  "log"

)

func main() {

  _, err := profile.NewProfileManager1()
  if err != nil {
  	log.Fatal("error: ", err)
  }

}
```

Results in:
```
error: Properties.GetAll org.bluez.ProfileManager1: Method "GetAll" with signature "s" on interface "org.freedesktop.DBus.Properties" doesn't exist

```

I've had no issues implementing similar code in Python. Any ideas what I'm doing wrong?

muka, 2019-08-19:
Hi
there was an issue in the generated code, properties are ignored now. Let  me know.

I never used that part of the API, would you mind to provide and example usage so I can add it to  the library?

Thank you

jaymell, 2019-08-20:
Wow. Thank you for the quick fix! So far things are looking good. I've got the profile registered successfully. Essentially I'm just trying to set up a virtual BT keyboard based largely off this python repo: https://github.com/mlabviet/BL_keyboard_RPI. 

I need something more scaleable than a python service, though, hence the attempt to do this in go. Luckily I found your library!

Thanks!

muka, 2019-08-20:
That would be great, thank you

I am closing the issue, please reopen if needed

closed 2019-08-20 by muka

SerialVelocity, 2023-12-02:
@jaymell Did you end up uploading the code for setting up a virtual BT keyboard in Go anywhere? I've been trying to do it recently (I started with other libraries but they were all missing the necessary features)