muka/go-bluetooth #85: build failed ： undefined xxx

ZetaChow, 2019-09-05:
● ~ » cd $GOPATH/src/github.com/muka/go-bluetooth && make build                         
CGO_ENABLED=0 go build -o go-bluetooth ./main.go
# github.com/muka/go-bluetooth/hw/linux/hci
hw/linux/hci/hci.go:7:9: undefined: socket.Down
hw/linux/hci/hci.go:12:9: undefined: socket.Up
hw/linux/hci/hci.go:17:9: undefined: socket.List

muka, 2019-09-05:
Do you have dependencies installed?

Try with `go get -u -v ./...`

ZetaChow, 2019-09-05:
I try `go get -u -v ./...`
other error show:
bluez/profile/network/gen_Network1.go:290:50: syntax error: unexpected upon, expecting comma or )
bluez/profile/network/gen_Network1.go:292:20: syntax error: unexpected either at end of statement

muka, 2019-09-05:
It was a bug. I will fix the generation but in the meanwhile this commit should fix it

muka, 2019-09-13:
Closing, let me know if the issue is still there. Thank you

closed 2019-09-13 by muka