muka/go-bluetooth #86: Pairing request different from bluetoothctl

rtchoboian, 2019-09-26:
Hello,

I try to to establish a secure connection with a device, but my pairing request always fails through go-bluetooth but works thanks to Bluetoothctl.
I have analyzed the differences thanks to Btmon and I found that the pairing requestes have not the sames parameters.
Do you know how I can configure my adapter through go-bluetooth to obtain the same request as through Bluetoothctl?
Thank you

### Request through Bluetootctl
_< ACL Data TX: Handle 3585 flags 0x00 dlen 11                                                                                                                                                  #17 [hci0] 3.894838
      SMP: Pairing Request (0x01) len 6
        IO capability: **KeyboardDisplay (0x04)**
        OOB data: Authentication data not present (0x00)
        Authentication requirement: **Bonding**, MITM, SC, No Keypresses, CT2 (0x2d)
        Max encryption key size: 16
        Initiator key distribution: **EncKey Sign LinkKey (0x0d)**
        Responder key distribution: **EncKey IdKey Sign LinkKey (0x0f)**_

### Request through go-bluetooth
_< ACL Data TX: Handle 3585 flags 0x00 dlen 11                                                                                                                                                  #44 [hci0] 8.494573
      SMP: Pairing Request (0x01) len 6
        IO capability: **NoInputNoOutput (0x03)**
        OOB data: Authentication data not present (0x00)
        Authentication requirement: **No bonding, No MITM**, SC, No Keypresses, CT2 (0x28)
        Max encryption key size: 16
        Initiator key distribution: **none (0x08)**
        Responder key distribution: **IdKey LinkKey (0x0a)**_

rtchoboian, 2019-10-01:
I have found how to change the _IO  capability_ (_agent.ExposeAgent(conn, ag, agent.CapKeyboardDisplay, true)_)
But I still don't know how to change the _Authentication requirement_ from **No bonding** to **Bonding**.

Thanks to _btmon_, I see that the first command that my go programm send is:
_@ MGMT Command: Set Bondable (0x0009) plen 1                                                                                                                                             {0x0001} [hci0] 14.792767
        Bondable: Disabled (0x00)_

Do you know how I can disable this command?

Thank you

mattwallington, 2020-04-02:
@rtchoboian Did you have any luck with this?

muka, 2020-04-02:
Maybe this can be achieved via the "so-called Bluetooth Management sockets"
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/mgmt-api.txt

There are wrappers to command-line tools, such as [btmgmt](https://github.com/muka/go-bluetooth/blob/master/hw/linux/btmgmt/btmgmt.go#L202)

Does this help?

muka, 2020-07-28:
I do not have a solution, it relates probably to bluez configurations. Please reopen if still needed

closed 2020-07-28 by muka