muka/go-bluetooth #87: custom service and chars issue

freedreamer82, 2019-11-28:
hello,

i've been trying to setup a custom service with 2 chars but following your example and connecting eith my phone  I see a service but without chars.

below the code
```

s1, err := a.NewService()
	if err != nil {
		return err
	}

	c1, err := s1.NewChar()
	if err != nil {
		return err
	}

	err = s1.AddChar(c1)
	if err != nil {
		return err
	}

	c1.OnRead(service.CharReadCallback(func(c *service.Char, options map[string]interface{}) ([]byte, error) {
		log.Warnf("GOT READ REQUEST")
		return []byte{42}, nil
	}))
	c1.OnWrite(service.CharWriteCallback(func(c *service.Char, value []byte) ([]byte, error) {
		log.Warnf("GOT WRITE REQUEST")
		return value, nil
	}))

	d1, err := c1.NewDescr()
	if err != nil {
		return err
	}

	err = c1.AddDescr(d1)
	if err != nil {
		return err
	}

	err = a.AddService(s1)
	if err != nil {
		return err
	}

	err = a.Run()
	if err != nil {
		return err
	}
```

moreover if I want to customize a new service app with my UUID how should I do ? from the doc is not clear...

thanks for the great work!

Marco

muka, 2019-11-28:
Hi Marco
thank you for the issue. It's a bit I am not working on the repo, I need to give check if there is any bug or incosistency with bluez interfaces.

There is also an example folder with some more details

freedreamer82, 2019-11-29:
Hi Luca,

I really like the project and it is what i need.Could you please double check if does it work on your side?
Moreover the service interface to me should be changed in order to simplify the creation of a  custom service (maybe passing the UUID in the NewMethod for example).

if the project really does what I need I can contribute with PR if necessary.

Thanks

              Marco

muka, 2019-11-29:
If you have the time to work on a PR it would be great! 

Let me know if I can provide support somehow

freedreamer82, 2019-11-29:
Of course I can work on a PR.
BTW I would start on it if i'm sure the lib can do what I need.If you can check your example and get it to  work I'll modify the service trying to make it more generic.
Right now i'm using paypal lib and works pretty well but it's not integrated with bluez dbus and I would like to use bluetoothd .

thanks again

Marco

mattwallington, 2020-04-02:
I am also having this issue.  I can create custom services but the characteristics and descriptors don't show up.

muka, 2020-04-02:
Hi @freedreamer82  @mattwallington I will try to sort this out. I have the same issue

mattwallington, 2020-04-02:
Much appreciated!

mattwallington, 2020-04-07:
@muka I have been using Bluez 5.50.  Do you think the issue might be a spec change or something?  Perhaps rolling back to a previous Bluez version would solve the issues?  If so, do you have a version you would recommend that is known to be working?  Thanks so much for this library and all the work you've put into it.

freedreamer82, 2020-04-11:
hi, news?

muka, 2020-04-12:
Hello, an Eastern update that surely will drive us in the right direction..

I refactored a bit the generation code and added a docker based build that should simplify testing across bluez versions (branch [bluez-5.54](https://github.com/muka/go-bluetooth/tree/bluez-5.54)).

I am testing with 5.54 to see if there are differences. Additionally, I will rebuild the service to be aligned to the  python example in the bluez repository. I'd expect the interoperability of the two example and to spot the bug(s) that way. :tada:

muka, 2020-04-13:
Good news, the service seems to work now. It was missing the `btmgmt le on` command.

I will work more on the examples so that the example client read, write and subscribe to the example service

freedreamer82, 2020-04-13:
I ll try it. btw remains the issue to customize uuid of the customer service.could u Expose a method?

muka, 2020-04-13:
Sure I will, if you want to provide a PR while testing I will be happy to merge it. Thanks

mattwallington, 2020-04-13:
Awesome news.  Thank you for fixing that!

freedreamer82, 2020-04-14:
I tried but i have the same issue with my iphone and lightblue app. It see the device but no services and chars.

already did a "btmgmt le on"

I'm on ubuntu bluez 5.50

muka, 2020-04-14:
Ok, I have the suspect 5.50 is not really playing nice with services (as @mattwallington suggested)

Can you try to run on bluez 5.54 ?

An example (requires docker):

```sh
cd <path>/go-bluetooth
git checkout master && git pull
make docker/bluetoothd/run
./bin/docker-btmgmt -i hci0 le on
./bin/docker-btmgmt -i hci0 bredr off
./bin/docker-btmgmt -i hci0 power on
go run main.go service server --adapterID hci0
```
Service is exposed and I can see the descriptor in d-feet.

I will spend more time on it, any help is appreciated
Thanks

freedreamer82, 2020-04-14:
I can try... but with python it works well on dbus and 5.50 so I guess it's not an bluez issue.

M.

freedreamer82, 2020-04-14:
/org/bluez/hci0/apps/app0/service_000003e8/char0/descr0

for instance this is NOT exposed on dbus...

muka, 2020-04-14:
Thank you @freedreamer82! 
You are right, chars and descr where not exposed in the ObjectManager. 

We have made a step forward  but still I cannot see the list of chars / descr..

There is an error in the logs that I am investigating 

```
bluetoothd[13557]: profiles/gap/gas.c:handle_characteristic() Unsupported characteristic: 00002aa6-0000-1000-8000-00805f9b34fb
```

As usual, suggestion are welcome!

freedreamer82, 2020-04-15:
I tried 2 debug.
First of all you need a conf file in /etc/dbus-1/system.d/go-bluetooth.conf
something like this

<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-BUS Bus                             
Configuration 1.0//EN"                                                                
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">                       
<busconfig>                                                                           
  <policy context="default">
    <allow own="go.bluetooth"/>
    <allow send_destination="go.bluetooth"
        send_interface="org.freedesktop.DBus.Introspectable"/>
  </policy>                                                                           
</busconfig>

after that there is another problem on Properties.GetAll org.bluez.Adapter1: Method "GetAll" with signature "s" on interface "org.freedesktop.DBus.Properties" doesn't exist 

you're calling method on wrong interfaces

M.

freedreamer82, 2020-04-15:
if can help I have a python bluex example running and it exposes

https://kernel.googlesource.com/pub/scm/bluetooth/bluez/+/5.36/test/example-gatt-server
└─/org
  └─/org/bluez
    └─/org/bluez/example
      ├─/org/bluez/example/advertisement0
      └─/org/bluez/example/service0
        ├─/org/bluez/example/service0/char0
        └─/org/bluez/example/service0/char1

muka, 2020-04-15:
Hi @freedreamer82, 
the repository has example dbus configuration in [./env/dbus](https://github.com/muka/go-bluetooth/tree/master/env/dbus) and the install procedure is in the [readme](https://github.com/muka/go-bluetooth#dbus-configuration-setup). 

I used to use the `org.bluez` namespace but it seemed to interfer with `dbus-send`, with the latest push the interfaces are also browsable via d-feet so I do not think the problem is there.

[image]

I am reviewing how the properties are exposed, in particular for characteristics.

freedreamer82, 2020-04-16:
sorry didn't see but I already have them. BTW same error... moreover also your makefile doesn't work
on master
cd src/bluez && git fetch && git checkout 5.50
error: pathspec '5.50' did not match any file(s) known to git
make: *** [Makefile:23: bluez/checkout] Error 1

muka, 2020-04-16:
Ok right, it misses the bluez submodule `make bluez/init`. 

I have used BLE Scanner on my phone and it allow me to R/W char and descriptors.
Can you share details of your setup (server code / client)?

Thanks

freedreamer82, 2020-04-16:
so .. I remove the folder ..clone it again..installed with the submodule...it goes ok.installed the conf file with the script but 

go run main.go service server --adapterID hci0
TRAC[0000] Exec: btmgmt [--index hci0 power off]        
TRAC[0000] Exec: btmgmt [--index hci0 le on]            
TRAC[0000] Exec: btmgmt [--index hci0 bredr off]        
TRAC[0000] Exec: btmgmt [--index hci0 power on]         
ERRO[0000] Error: Connection ":1.2314" is not allowed to own the service "go.bluetooth" due to security policies in the configuration file 
exit status 1

muka, 2020-04-16:
Seems the dbus policies are not enforced. I would check

1. the two files/symlink exists `ls /etc/dbus-1/system.d/dbus-go-bluetooth-*`  
2. dbus policies has been loaded by the daemon `make dev/dbus/reload` [source](https://github.com/muka/go-bluetooth/blob/master/Makefile#L54)
3. Your user has the `bluetooth` group `make docker/bluetoothd/init` [source](https://github.com/muka/go-bluetooth/blob/master/Makefile#L72)
4. [In doubt, reboot the machine to apply above updates]

This reflect my own setup on ubuntu 19.10

freedreamer82, 2020-04-16:
ok , after a reboot it works :) thanks! I can close it

closed 2020-04-16 by freedreamer82

muka, 2020-04-16:
thank you for the help

mattwallington, 2020-04-17:
Nice work!

aykevl, 2020-06-15:
This error should now be fixed with #105, without requiring any system configuration changes:

```
Connection ":1.2314" is not allowed to own the service "go.bluetooth" due to security policies in the configuration file
```