muka/go-bluetooth #91: Deadlock when receiving GATT notifications and remote device disconnects

Depau, 2020-01-25:
```golang
var char gatt.GattCharacteristic1 = GetMyGattCharacteristic1()
if chan, err = char.WatchProperties(); err != nil {
    return
}
if err := char.StartNotify(); err != nil {
	return
}

// do work
// remote device dies

```

muka, 2020-01-25:
Hi @Depau 
I have a dev branch which is mostly updated but poorly tested yet. If you want to give it a shot it is `app_refactor`.

I think I broke the `service` exposing implementation, but the rest should be up to date

Depau, 2020-01-26:
Hi,
I tried with the `app_refactor` branch but the deadlock is still occurring. I'll try to see why in the next days.

muka, 2020-01-26:
Thanks, that branch is merged into master now. Please ensure to run `make all` as the library need to generate code from the bluez API docs.

muka, 2020-03-31:
Closing, please reopen if  needed.

closed 2020-03-31 by muka