muka/go-bluetooth #92: Missing module for example

robbat2, 2020-02-02:
`github.com/muka/go-bluetooth/bluez/profile/battery` is referenced in `examples/`, but does not seem to exist in the tree presently. This makes `go get` fail.

```
$ go get -v github.com/muka/go-bluetooth
github.com/muka/go-bluetooth (download)
package github.com/muka/go-bluetooth/bluez/profile/battery: cannot find package "github.com/muka/go-bluetooth/bluez/profile/battery" in any of:
        /usr/lib/go/src/github.com/muka/go-bluetooth/bluez/profile/battery (from $GOROOT)
        /home/robbat2/go/src/github.com/muka/go-bluetooth/bluez/profile/battery (from $GOPATH)
```

muka, 2020-02-02:
I need to fix this thanks, in the meanwhile you can move to the library
directory and run 'make all'

sent from my phone

muka, 2020-03-31:
Hi, I updated the `README` to cover how to get setup working

Basically, clone the repo and run `make`.

Closing, please reopen if needed
Thanks

closed 2020-03-31 by muka