muka/go-bluetooth #93: Broken in new Go versions (1.13+)

natdm, 2020-03-02:
Uses an out of date version of Logrus, which uses an out of date `/x/unix` import from Go, resulting in the build error:

```bash
omit@omit-MacBook-Pro go-bluetooth (master) $ make build
git submodule update
FILTER= go run gen/srcgen/main.go
# github.com/sirupsen/logrus
../../sirupsen/logrus/terminal_check_bsd.go:11:12: undefined: unix.IoctlGetTermios
make: *** [gen/run] Error 2
```

muka, 2020-03-02:
Thank you for reporting this. I will fix it asap, if you can provide a PR I
will revise and merge

muka, 2020-03-31:
Hi, 
I reviewed and updated the dependencies. I also never tried the bluez on mac, I am unsure if it even works.

Closing, please reopen if needed
Thanks

closed 2020-03-31 by muka