muka/go-bluetooth #94: Connect with device without pairing or at least automatically pair

n0he, 2020-03-02:
Hi,

I'm a bit overwhelmed by the code, so instead of just browsing through it, not knowing what to look for exactly, I decided to just ask here. Is there a way to connect to a device as client without pairing or not having to confirm the pin popup-message?
Thanks for any Help

muka, 2020-03-02:
Technically yes, as long your device allow this.

n0he, 2020-03-09:
I realized that I actually wouldn't want to pair but just connect. Is this possible? I am using "nRF Connect" on my Android phone to advertise some services and I can connect without Pairing from other Phones. I would like the same behaviour with go-bluetooth, but don't seem to figure out how to do that, because the example provided in "examples/service/client.go" uses "dev.Pair()". I thought this example was a BLE GATT server/client example, is that correct? If so why is "Pair()" needed and is there a way to avoid it?
Thank you for your help either way!

muka, 2020-03-31:
I tried to search a bit and seems pairing may be not required, depending on your setup.

My suggestion is to try from the bluetooth command line, see this reference from [ubuntu core](https://docs.ubuntu.com/core/en/stacks/bluetooth/bluez/docs/reference/pairing/introduction)

If you find a way, you can replicate it via the bluez API thus this library.

Closing, please reopen if needed!

closed 2020-03-31 by muka