muka/go-bluetooth #95: How to include into another project as module

Depau, 2020-03-31:
Following up from #91 since it's a different issue.

I see that you merged the patches I was interested in, but since now it is necessary to generate the wrappers with `make`, one cannot `go get` go-bluetooth any more or import as module as I'd normally do with libraries.

How are users supposed to import it?

muka, 2020-04-01:
Hi, good point.. In theory the code generation should happen when the module is installed. This seems not to be the case..

I have tried asking in the golang channels, will follow up soon. Do you have any suggestion?

The relevant part is in [main.go](https://github.com/muka/go-bluetooth/blob/master/main.go#L1)

Depau, 2020-04-01:
Hi, I wasn't aware of `go generate`, that's cool.

However I see that it's usually used in "final" programs, not libraries that are supposed to be imported.

While waiting for better suggestions from people in the golang channels, my only suggestion is to have a separate branch that is never pushed to manually, but is generated by a CI based on another branch. It's definitely ugly, I know, but this way `go get` should work.

muka, 2020-04-01:
I seen other projects doing like that, I think is just me doing it wrong. If it is not the case, the CI approach would be definitely an option!

Thanks

muka, 2020-04-11:
This should be fixed now, please reopen if needed. Thanks !

closed 2020-04-12 by muka

Depau, 2020-04-15:
Great, thank you!

Also +1 for writing the only Bluez library I could find (for basically every programming language I looked for) that actually uses the DBus API instead of calling the commands.

IDK what's wrong with people, the DBus API is actually quite simple :/