muka/go-bluetooth #98: Advertising data too long or couldn't be generated.

Edwin-Luijten, 2020-04-21:
I fail to understand why I get below error.

Running below code gives this error: ```2020/04/21 09:04:01 Failed to parse advertisement.```  
```go
b, err := beacon.CreateEddystoneUID("EDD1EBEAC04E5DEFA017", "0BDB87539B67", -59)
if err != nil {
	log.Fatal(err)
}

timeout := uint16(60 * 60 * 18)

cancel, err := b.b.Expose("hci0", timeout)
if err != nil {
	log.Fatal(err)
}

defer cancel()

select {}
```

btmon output:
```
Bluetooth monitor ver 5.50
= Note: Linux version 5.3.0-7648-generic (x86_64)                                                  0.634700
= Note: Bluetooth subsystem version 2.22                                                           0.634703
= New Index: 9C:B6:D0:97:A6:48 (Primary,USB,hci0)                                           [hci0] 0.634704
= Open Index: 9C:B6:D0:97:A6:48                                                             [hci0] 0.634704
= Index Info: 9C:B6:D0:97:A6:48 (Qualcomm)                                                  [hci0] 0.634705
@ MGMT Open: bluetoothd (privileged) version 1.14                                         {0x0001} 0.634706
@ MGMT Open: btmon (privileged) version 1.14                                              {0x0002} 0.634731
@ MGMT Command: Set Connectable (0x0007) plen 1                                    {0x0001} [hci0] 3.603181
        Connectable: Enabled (0x01)
@ MGMT Event: Command Complete (0x0001) plen 7                                     {0x0001} [hci0] 3.603197
      Set Connectable (0x0007) plen 4
        Status: Success (0x00)
        Current settings: 0x00000adb
          Powered
          Connectable
          Discoverable
          Bondable
          Secure Simple Pairing
          BR/EDR
          Low Energy
          Secure Connections
@ MGMT Event: New Settings (0x0006) plen 4                                         {0x0002} [hci0] 3.603199
        Current settings: 0x00000adb
          Powered
          Connectable
          Discoverable
          Bondable
          Secure Simple Pairing
          BR/EDR
          Low Energy
          Secure Connections
@ MGMT Command: Set Discoverable (0x0006) plen 3                                   {0x0001} [hci0] 3.603270
        Discoverable: General (0x01)
        Timeout: 64800
@ MGMT Event: Command Complete (0x0001) plen 7                                     {0x0001} [hci0] 3.603273
      Set Discoverable (0x0006) plen 4
        Status: Success (0x00)
        Current settings: 0x00000adb
          Powered
          Connectable
          Discoverable
          Bondable
          Secure Simple Pairing
          BR/EDR
          Low Energy
          Secure Connections
= bluetoothd: Advertising data too long or couldn't be generated.                                  3.607407
```

muka, 2020-05-05:
Hi @Edwin-Luijten I will dig deeper as the bluetoothd log is not very descriptive as you mentioned. Let me know if you find anything.

muka, 2020-07-28:
This should work now, I have added it to examples too

```sh
cd _examples
go run main.go beacon eddystone UID
```

commit 82a7b3d9 (2020-07-28): remove service UUID from adverting, for eddystone UID length. Fix #98
closed 2020-07-28 by muka