muka/go-bluetooth #102: Properties.GetAll org.bluez.Battery1: No such interface 'org.bluez.Battery1'"

freedreamer82, 2020-05-13:
Hi Luca,

I've been using your lib in client mode,  when I correctly connect to a device and suddenly request battery profile i have this error:

Properties.GetAll org.bluez.Battery1: No such interface 'org.bluez.Battery1'"

if a put a delay of 1 sec it works properly. I guess it's not yet registered to dbus , maybe it can be useful to wait the signal

what u think ?

       Marco

muka, 2020-07-28:
Hi Marco
what you describe is definitely useful but would require to manage states with some sort of events. I do not have time to implement new feature but PR are always welcome!

muka, 2020-12-11:
Closing, please reopen if needed. Thanks!

closed 2020-12-11 by muka