muka/go-bluetooth #108: deadlock in dbus registration

shigmas, 2020-06-01:
I've hit two deadlocks in running the discovery example. I think they may be the same cause. Or, in the very least, it might be the same steps that can be used to avoid the deadlock, since I'm very new to the bluez/dbus stack.

Anyway, It's easier to read in code, so I created a PR:
https://github.com/muka/go-bluetooth/pull/107

muka, 2020-06-01:
Hi, thank you for taking the time to report. I will check it, if you come to a solution please finalize your PR so we can review and merge.

muka, 2020-07-29:
Hi I am reviewing the issue you reported. 

I do agree the `handleBeacon` call block the context, will attempt a fix. The easiest is to ensure a context is avail for it to be cancelled and also run into a goroutine.

Let's see if the deadlock persist after those changes and from where it comes

Thank you!

closed 2020-12-11 by muka

shigmas, 2020-12-11:
Was this fixed?