muka/go-bluetooth #111: how to scan all ble devices

vtolstov, 2020-06-28:
I'm run bluetoothd with -E flag (experimental features enabled) and have lego boost and honor band 4 and want to discover them.
But discover finds only raspberry pi 4 and nothing more.

mteodor, 2020-07-18:
I'm experiencing similar problem, when I run discovery example I only got one device, and If i turned that device off the other one would appear but always one.
However when I do scan with bluetoothctl I can se several ble devices
Running on ubnutu 20.04
bluetoothd[47660]: Bluetooth daemon 5.53
bluez 5.53

muka, 2020-07-28:
Hi, thank you for reporting (and sorry for the late response).

This is related to #108, will investigate further

muka, 2020-07-30:
With the last commit you should start seeing more devices, seems there is still a dead-lock issue to fix

muka, 2020-09-29:
Should be fixed, please reopen if needed. Thanks

closed 2020-09-29 by muka