muka/go-bluetooth #112: Installing bluetooth dbus permission configuration bricks linux install

IoTPanic, 2020-07-20:
Hey all, I tried to run the example and since it was having issues running I tried to complete the DBus configuration setup and right after running  the `make dev/dbus/install` while in the root project directory on Fedora 31 my screen went black with a flashing cursor, and restarting it will cause it to boot into the same. There is no function button that will help, and I never make it to the GUI, so unless you have a quick fix its lucky BTRFS took regular snapshots.

Any suggestions for resolving in mind?

muka, 2020-07-28:
Hi, very sorry that this happened! 
Thank you for reporting anyway, I would suggest to open a terminal (usually with `ctrl+alt+f1`) and run `ls /etc/dbus-1/system.d/ | grep dbus-go`

Here you will have listed the 2 files installed with the make command. You can remove them and reboot the system, it should be enough to avoid mis-configuration with dbus.

Also if you have evidence of what happened or can solve the issue with the script or configuration it would be a great addition to the project. Your contribution can prevent problems to others

Thank you

IoTPanic, 2020-07-28:
Hey @muka,

No worries, I tried my hardest to get a terminal back so I could attempt getting more info and recovery but I could not, I don't think I used `ctrl+alt+f1` but I tried a few other key combos and tried to get a shell from grub but due to the fact I needed a working computer ASAP that day, had a recent file backup, and was going to hop distros soon anyways, I just reinstalled an OS. If I can make time in the next few days I will try to recreate this issue in a VM or on bare metal and see what I can do to fix the issue.

Thanks,
Samuel

muka, 2020-09-07:
Dropped from the documentation as the code works also without those specific permissions. Thanks!

closed 2020-09-07 by muka