muka/go-bluetooth #113: Detacched gorotine in WatchProperties

1am, 2020-08-11:
Hello,

I've got a question about the go routine launched in https://github.com/muka/go-bluetooth/blob/master/bluez/props.go#L20 when `WatchProperties` is called. In my case I'm periodically connecting to a device, start listening for notifications, send some data, wait for response and disconnect. The subscription to notifications is practically identical as in https://github.com/muka/go-bluetooth/blob/master/devices/sensortag/luxometer.go#L163 which is pretty straight forward.

But I've noticed over time that my memory usage grows over time. I started digging through my code, cutting away pieces and eventually noticed that when i call
```
d.Connect()
ch := char.WatchProperties()
time.Sleep(1 * time.Seconds)
char.UnwatchProperties(ch)
d.Disconnect()
```
Within each cycle I'm just printing the number of go routines with:

```
num := runtime.NumGoroutine()
log.Debugf("NumGoroutine = %d", num)
```

I always end up with one more go routine more than I started with. While looking through the go-bluetooth API and examples I started wondering about the line https://github.com/muka/go-bluetooth/blob/master/bluez/props.go#L53. I'm not sure if I'm correct but the loop will be stuck there forever waiting for incoming data (which will not happen) and never iterate on the outter loop which would handle closing of the channel correctly. It seems very consistent that just calling `WatchProperties` and later `UnwatchProperties` always leaves me with one more routine than I started with but I can't really pin down the reason for it except for my thinking about the mentioned loop. 

Is that right or I'm looking in the wrong place? Because during writing this message my go routine count grew by 80. If I comment out the `WatchProperties` it will stay constant.

muka, 2020-08-11:
Hello thank you for the detailed write-up. I think you are hitting a bug that has been lurking in the library but I have not been able to fix yet!

Can you provide a PR to fast thing up?

I suspect this issue would related also to other issues open

Thanks!

1am, 2020-08-11:
Hello. Tried writing a test but so far failed on making a mocked GattCharacteristic1 instance to make a test which would isolate the issue. Maybe you have a way to do that? Looked through the code but couldn't find it.

muka, 2020-08-11:
Ahw testing has been a long standing desiderata for this project.. The best I come up is to use a physical device like TI sensor tag or on Android device.

You can check the server examples where there is a client server interaction, but it requires two adapters

Bluez project has python implementation also under test/

If you have any better idea I am very happy to know

Thanks

1am, 2020-08-13:
I've been playing around the problem recently and don't have super reproducable test but adding `wprop.GetWatchPropertiesChannel() <- nil` above https://github.com/muka/go-bluetooth/blob/master/bluez/props.go#L98 reduces the number of leftover go routines from 2 to 1 which kind of makes sense. What I'm trying to understand is where the other 1 comes from.

Not making a pull request yet as this is an one change. I have a simple test script which could probably help wit replicating but before I push I wanted to ask how do you make the following code execute without running a scan on the adapter first?

```
a, err := api.GetAdapter(adapterID)
dev, err := a.GetDeviceByAddress(address)
```

The pasted fragment comes from https://github.com/muka/go-bluetooth/blob/master/examples/sensortag_info/sensortag_info.go#L24 I cannot do the same, always have to run a short scan to populate the list of devices. It's not a problem but I'm wondering how to do it otherwise.

muka, 2020-08-14:
Thank you for the update, bluez keep track of scanned device in a cache. If the device is already in the list the method should find it by address. Ensure you are not flushing the cache every time.

Looking forward to integrate your findings!

closed 2020-08-15 by 1am