muka/go-bluetooth #118: go run main.go not working

diaojunxian, 2020-08-27:
As I want to try the _examples/main.go，and clone the code in macOS 10.15.5 (19F101),

the the go version is `go version go1.15 darwin/amd64`

try the command 
```
cd _examples

go run main.go
```

it reports that
```
# github.com/muka/go-bluetooth/hw/linux/hci
../hw/linux/hci/hci.go:67:25: undefined: unix.AF_BLUETOOTH
../hw/linux/hci/hci.go:67:59: undefined: unix.BTPROTO_HCI
../hw/linux/hci/hci.go:107:8: undefined: unix.SockaddrHCI
../hw/linux/hci/hci.go:171:25: undefined: unix.AF_BLUETOOTH
../hw/linux/hci/hci.go:171:59: undefined: unix.BTPROTO_HCI
../hw/linux/hci/hci.go:184:25: undefined: unix.AF_BLUETOOTH
../hw/linux/hci/hci.go:184:59: undefined: unix.BTPROTO_HCI
../hw/linux/hci/hci.go:200:25: undefined: unix.AF_BLUETOOTH
../hw/linux/hci/hci.go:200:59: undefined: unix.BTPROTO_HCI
```

muka, 2020-08-27:
Hi, this library works with linux

diaojunxian, 2020-08-27:
@muka ok, thanks a lot, the how can I run it in macOS？

muka, 2020-08-27:
Not sure, I do not have a mac to test. You may have luck using a VM or docker with a linux base image such as ubuntu. Let me know if you can make it run so we can update the README with some documentation.

Thank you!

wiltongarcia, 2020-09-04:
Hi @muka I'm running it on Ubuntu Mate in a Raspberry PI 4 and it is failing:  

```bash
$ go run main.go 
# github.com/muka/go-bluetooth/bluez/profile/obex
../bluez/profile/obex/gen_MessageAccess1.go:316:88: undefined: Message
../bluez/profile/obex/gen_MessageAccess1.go:318:13: undefined: Message
../bluez/profile/obex/gen_PhonebookAccess1.go:412:68: undefined: VCardItem
../bluez/profile/obex/gen_PhonebookAccess1.go:414:13: undefined: VCardItem
../bluez/profile/obex/gen_PhonebookAccess1.go:458:98: undefined: VCardItem
../bluez/profile/obex/gen_PhonebookAccess1.go:460:13: undefined: VCardItem
```

During the code generation it returned some warnings:
```bash
$ BLUEZ_VERSION=5.54 make bluez/init bluez/checkout gen/clean gen/run
git submodule init
git submodule update
cd src/bluez && git fetch && git checkout 5.54
Previous HEAD position was 152449948 Release 5.53
HEAD is now at ab8a80b04 Release 5.54
rm -f `ls bluez/profile/*/gen_* -1` || true
BLUEZ_VERSION=5.54 FILTER= go run gen/srcgen/main.go full
INFO[0000] API 5.54                                     
WARN[0000] Return type value is too long? `array{objects, properties}` 
WARN[0000] Return type value is too long? `object node, array{byte, array{(uint16, dict)}} configuration` 
WARN[0000] Return type value is too long? `uint16 net_index, uint16 unicast` 
WARN[0000] Return type value is too long? `array{string vcard, string name}` 
WARN[0000] Return type value is too long? `array{string vcard, string name}` 
INFO[0000] Saving to ./bluez-5.54.json                  
INFO[0000] Generating from ./bluez-5.54.json
```

muka, 2020-09-05:
Thank you for reporting.
I have worked on the generated part recently. Try fetching the repository.

Also branch bluez/5.54 should be already usable

Let me know

commit fd9df0dd (2020-09-07): added missing types definition. Fixes #118
closed 2020-09-07 by muka