muka/go-bluetooth #119: In the discovery event it is possible to carry out the filters

addodelgrossi, 2020-08-31:
Can we filter only devices that have a specific Service?

Or should I always connect and list services?

muka, 2020-08-31:
Hello, yes it is provieded by bluez with `SetDiscoveryFilter` please refer to the [bluez docs](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/adapter-api.txt#n48) for more details

addodelgrossi, 2020-08-31:
Thanks for reply !!

Can I set directly on the adapter?

I did but it is not filtering.

```golang
	log.Debug("Start discovery")
	filterMap := make(map[string]interface{})
	filterMap["UUIDs"] = []string{"05e2b551-4589-4449-b45e-195c21328568"}
	a.SetDiscoveryFilter(filterMap)
	discovery, cancel, err := api.Discover(a, nil)
```

muka, 2020-09-02:
Not sure, try looking into the bluetoothd logs, debug can be enabled with -d flag.

If you have proof it's a bug we can look in more detail at it, thanks

addodelgrossi, 2020-09-02:
Thank you ! I will analyze the log and if I find something I answer

muka, 2020-12-11:
Closing, please reopen if needed

closed 2020-12-11 by muka