muka/go-bluetooth #125: Non BLE devices, Bluetooth HID/L2CAP

jackjameshoward, 2020-09-25:
I know this isn't an issue, but it's hard to find another place to open a discussion about this.
I've been digging through Bluez and go-bluetooth for a while now, and I'm not winning. 

I want to be able to read and request data from an old Wiimote.
These aren't the BLE devices but appear as HID devices.
https://wiibrew.org/wiki/Wiimote#HID_Interface
I can pair/connect to it, but that's about it.
I've been through the bluez repo [here](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/profiles/input/device.h) and found an input profile that defines the same control and data pipe as described in the wiki Wiimote#HID_Interface linked above.
```
#define L2CAP_PSM_HIDP_CTRL	0x11
#define L2CAP_PSM_HIDP_INTR	0x13
```
I'd like to know if it is possible to send requests and read responses using go-bluetooth? 
If it is possible, do you have any guidance for me how to achieve this? 
Does go-bluetooth implement most of the Bluez functions? 
Much appreciated in advance.

muka, 2020-09-28:
Hi, I am not very experienced on this topic, sorry.
To understand the topic I found this discussion https://stackoverflow.com/questions/22688375/writing-hid-service-with-pybluez

If it can be done over dbus -ideally- this library may be able to address it. 

Let me know, thanks!

muka, 2020-12-11:
Closing, please reopen if needed!

closed 2020-12-11 by muka