muka/go-bluetooth #131: [Question] How to get non null characteristic handles?

Braincoke, 2021-03-21:
Hi,

Thanks for your work, I've just discovered this library and I spent quite some time trying to make it work.

## Issue

I have setup an ESP-32 with [ble-ctf](https://github.com/hackgnar/ble_ctf) and I am trying to use go-bluetooth to solve the CTF (to get acquainted with the library).
I managed to find my device, connect to it and get a list of characteristics but my handles are always equal to 0.

## Code

Here is the gist of how I am using the library. I have a fully "working" snippet [here](https://gist.github.com/Braincoke/41ea40a13f8e0f56f6e20c892218a630)

```go
	// Display device characteristics
	chars, err := dev.GetCharacteristics()
	if err != nil {
		log.Error(err)
		os.Exit(5)
	}
	log.Infof("Found %d characteristics", len(chars))

	for _, char := range chars {
		// Uncomment to get the value
		// p := map[string]interface{}{}
		// char.ReadValue(p)
		charProps, _ := char.GetProperties()
		handle := charProps.Handle
		value := charProps.Value
		uuid := charProps.UUID

		// ISSUE : handle is always 0
		log.Infof("handle = %d, uuid = %s , value = %s", handle, uuid, hex.EncodeToString(value))
	}
```

I don't have the issue with `gatttool`:

```bash
gatttool -i hci0 -b B8:F0:09:C5:B5:2E --characteristics
handle = 0x0002, char properties = 0x20, char value handle = 0x0003, uuid = 00002a05-0000-1000-8000-00805f9b34fb
handle = 0x0015, char properties = 0x02, char value handle = 0x0016, uuid = 00002a00-0000-1000-8000-00805f9b34fb
```

What am I doing wrong? I have tried multiple variants like `char.GetProperty("Handle")` and so on but nothing works.

## System information

OS : Ubuntu 20.04
Bluez version : 5.53
Go version : 1.16.2

Chillimeat, 2021-12-16:
> Hi,
> 
> Thanks for your work, I've just discovered this library and I spent quite some time trying to make it work.
> I have the same problem. Do you have a solution？
> ## Issue
> I have setup an ESP-32 with [ble-ctf](https://github.com/hackgnar/ble_ctf) and I am trying to use go-bluetooth to solve the CTF (to get acquainted with the library). I managed to find my device, connect to it and get a list of characteristics but my handles are always equal to 0.
> 
> ## Code
> Here is the gist of how I am using the library. I have a fully "working" snippet [here](https://gist.github.com/Braincoke/41ea40a13f8e0f56f6e20c892218a630)
> 
> ```go
> 	// Display device characteristics
> 	chars, err := dev.GetCharacteristics()
> 	if err != nil {
> 		log.Error(err)
> 		os.Exit(5)
> 	}
> 	log.Infof("Found %d characteristics", len(chars))
> 
> 	for _, char := range chars {
> 		// Uncomment to get the value
> 		// p := map[string]interface{}{}
> 		// char.ReadValue(p)
> 		charProps, _ := char.GetProperties()
> 		handle := charProps.Handle
> 		value := charProps.Value
> 		uuid := charProps.UUID
> 
> 		// ISSUE : handle is always 0
> 		log.Infof("handle = %d, uuid = %s , value = %s", handle, uuid, hex.EncodeToString(value))
> 	}
> ```
> 
> I don't have the issue with `gatttool`:
> 
> ```shell
> gatttool -i hci0 -b B8:F0:09:C5:B5:2E --characteristics
> handle = 0x0002, char properties = 0x20, char value handle = 0x0003, uuid = 00002a05-0000-1000-8000-00805f9b34fb
> handle = 0x0015, char properties = 0x02, char value handle = 0x0016, uuid = 00002a00-0000-1000-8000-00805f9b34fb
> ```
> 
> What am I doing wrong? I have tried multiple variants like `char.GetProperty("Handle")` and so on but nothing works.
> 
> ## System information
> OS : Ubuntu 20.04 Bluez version : 5.53 Go version : 1.16.2

I have the same problem. Do you have a solution

> Hi,
> 
> Thanks for your work, I've just discovered this library and I spent quite some time trying to make it work.
> 
> ## Issue
> I have setup an ESP-32 with [ble-ctf](https://github.com/hackgnar/ble_ctf) and I am trying to use go-bluetooth to solve the CTF (to get acquainted with the library). I managed to find my device, connect to it and get a list of characteristics but my handles are always equal to 0.
> 
> ## Code
> Here is the gist of how I am using the library. I have a fully "working" snippet [here](https://gist.github.com/Braincoke/41ea40a13f8e0f56f6e20c892218a630)
> 
> ```go
> 	// Display device characteristics
> 	chars, err := dev.GetCharacteristics()
> 	if err != nil {
> 		log.Error(err)
> 		os.Exit(5)
> 	}
> 	log.Infof("Found %d characteristics", len(chars))
> 
> 	for _, char := range chars {
> 		// Uncomment to get the value
> 		// p := map[string]interface{}{}
> 		// char.ReadValue(p)
> 		charProps, _ := char.GetProperties()
> 		handle := charProps.Handle
> 		value := charProps.Value
> 		uuid := charProps.UUID
> 
> 		// ISSUE : handle is always 0
> 		log.Infof("handle = %d, uuid = %s , value = %s", handle, uuid, hex.EncodeToString(value))
> 	}
> ```
> 
> I don't have the issue with `gatttool`:
> 
> ```shell
> gatttool -i hci0 -b B8:F0:09:C5:B5:2E --characteristics
> handle = 0x0002, char properties = 0x20, char value handle = 0x0003, uuid = 00002a05-0000-1000-8000-00805f9b34fb
> handle = 0x0015, char properties = 0x02, char value handle = 0x0016, uuid = 00002a00-0000-1000-8000-00805f9b34fb
> ```
> 
> What am I doing wrong? I have tried multiple variants like `char.GetProperty("Handle")` and so on but nothing works.
> 
> ## System information
> OS : Ubuntu 20.04 Bluez version : 5.53 Go version : 1.16.2

Chillimeat, 2021-12-16:
I have the same problem. Do you have a solution？

muka, 2021-12-27:
I would suggest to check on dbus if it is actually returning the correct information. A tool like `dfeet` can help with a visual interface. If you find a solution would you like to share here so we can try to address with a fix?

Thank you

closed 2022-02-13 by muka