muka/go-bluetooth #134: btmgmt.GetAdapters panics in case of missing BT device

thierolm, 2021-05-06:
Hi, want to check, whether an BT device is connected and tried to use  
```go
devlist, err := hw.GetAdapters()
```
But it fails/panics, in case no BT device is connected:
```
panic: runtime error: slice bounds out of range [:6] with length 0 [recovered]
	panic: runtime error: slice bounds out of range [:6] with length 0
```
On linux CLI `btmgmt` shows a meaningful information ==> 0 items:
```
thierolm@tserver:/dev$ btmgmt info
Index list with 0 items
```
In the btmgmt.go module, this information in the first line is completely ignored: 
https://github.com/muka/go-bluetooth/blob/07f31c601d335ebe336c5923b2b66b2420d8e8aa/hw/linux/btmgmt/btmgmt.go#L57

Is it possible to return an empty BtAdapter array in such a case instead of raising a panic?

muka, 2021-05-07:
Thank you seems there is a bug there

Can you provide a PR with a fix?

thierolm, 2021-05-07:
> Can you provide a PR with a fix?

Sure, will try to provide one within the next view days. 👍

thierolm, 2021-05-07:
First insights: 
By testing all the flavors of GetAdapters functions (btmgmt info, hcitool dev, hciconfig) I realized, that the above error is not caused by btmgmt.go but by hciconfig.go.
Will focus on hciconfig.go ...
```
Running tool: /usr/local/go/bin/go test -timeout 30s -run ^TestGetAdapters$ github.com/muka/go-bluetooth/hw/linux/hciconfig

--- FAIL: TestGetAdapters (0.00s)
panic: runtime error: slice bounds out of range [:6] with length 0 [recovered]
	panic: runtime error: slice bounds out of range [:6] with length 0

goroutine 34 [running]:
testing.tRunner.func1.2(0x54d900, 0xc0001460d8)
	/usr/local/go/src/testing/testing.go:1143 +0x332
testing.tRunner.func1(0xc000106300)
	/usr/local/go/src/testing/testing.go:1146 +0x4b6
panic(0x54d900, 0xc0001460d8)
	/usr/local/go/src/runtime/panic.go:965 +0x1b9
github.com/muka/go-bluetooth/hw/linux/hciconfig.parseControllerInfo(0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, ...)
	/home/thierolm/evcc/git/go-bluetooth/hw/linux/hciconfig/hciconfig.go:62 +0x47d
github.com/muka/go-bluetooth/hw/linux/hciconfig.GetAdapters(0x15ffa9d0004d59b0, 0x609591fe, 0xc00010cf70, 0x485126, 0x609591fe)
	/home/thierolm/evcc/git/go-bluetooth/hw/linux/hciconfig/hciconfig.go:25 +0x14b
github.com/muka/go-bluetooth/hw/linux/hciconfig.TestGetAdapters(0xc000106300)
	/home/thierolm/evcc/git/go-bluetooth/hw/linux/hciconfig/hciconfig_test.go:41 +0x36
testing.tRunner(0xc000106300, 0x568ab8)
	/usr/local/go/src/testing/testing.go:1193 +0xef
created by testing.(*T).Run
	/usr/local/go/src/testing/testing.go:1238 +0x2b3
FAIL	github.com/muka/go-bluetooth/hw/linux/hciconfig	0.013s
FAIL
```