muka/go-bluetooth #137: Planned Support for SimpleAgent in App?

ahollist, 2021-05-28:
It looks like the `NewApp` function returns an `*App` that is initialized with a `SimpleAgent` that implements `Agent1Client`, which loses the capability to set the `PassKey` or `PassCode`. 

Is there any planned support for allowing setting the `PassKey/PassCode` in the App, or any plans for allowing setting a new `Agent` in the `App` that does allow setting those values? Or perhaps could I override the App's agent by creating a new `SimpleAgent` and exposing that after running the app? 

I know it's not generally a good practice to use a pregenerated key, but my particular device has no ability to display its key to the user when connecting and we do need a bit of additional security to not allow just anyone to connect to the device and write/read its characteristics. If there's another standard way to deal with this, I'd also be happy to do any additional reading you can point me to!

muka, 2021-05-29:
Hello @ahollist  I do not have a plan for this. Can you provide a PR to introduce that feature in the API? Happy to review and discuss.

Thanks

muka, 2021-12-27:
Closing, please reopen if you have updates. Thanks

closed 2021-12-27 by muka