muka/go-bluetooth #138: btmgmt provided no response

chemax, 2021-07-02:
Hi. When i'm triyng to get adapters list https://github.com/muka/go-bluetooth/blob/03c23c62f18121747785781f24d2a88f32eb7871/examples/btmgmt/btmgmt.go#L11
i have error "btmgmt provided no response". 
I'm use it on raspbian 10. How i can fix it?

muka, 2021-12-27:
Seems a setup issue to me, ensure btmgmt is working first. If you find a fix, please share a PR!

Closing, please reopen if you have updates. Thanks

ic2hrmk, 2022-01-08:
I can confirm that the issue still persists.

I tried to run the example you provide in the `example` directory, but found out that you have a very simple wrapper on top of it. So, I wrote a test to avoid using your wrapper method:
~~~go
package scratch

import (
	"bytes"
	"fmt"
	"log"
	"os/exec"
	"testing"
)

func TestBTMGMT(t *testing.T) {
	cmd := exec.Command("btmgmt", "info")
	outb, errb := new(bytes.Buffer), new(bytes.Buffer)
	cmd.Stdout = outb
	cmd.Stderr = errb
	err := cmd.Run()
	if err != nil {
		log.Fatal(err)
	}
	fmt.Println("out:", outb.String(), "err:", errb.String())
}
~~~
The result I've got:
~~~go
=== RUN   TestBTMGMT
out:  err: 
--- PASS: TestBTMGMT (0.01s)
PASS
~~~

This is something really weird with the `btmgmt`, it outputs the result of running it without the Go wrapper as it intended to.

My setup is:
~~~
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.2 LTS
Release:        20.04
Codename:       focal
Kernel: Linux scanner 5.15.6-051506-generic #202112010437 SMP Wed Dec 1 09:47:58 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
~~~

~~~
go version go1.17.3 linux/amd64
~~~

muka, 2022-01-08:
Thank you for reporting your tests, it is weird indeed. I suspect it being related to permission issues, eg. missing the cap `cap_net_admin`

Please consider using another approach, like hciconfig or bluetoothctl if it fit your needs

If you sort it out, a PR  is welcome!

ic2hrmk, 2022-01-09:
The approach with the `hciconfig` package worked fine for me, thanks!

~~~go
	hciInterface, err := hciconfig.GetAdapter(rcv.nic)
	if err != nil {
		return nil, nil, errors.Wrapf(err, "failed to acquire the bluetooth interface [hci=%s]", rcv.nic)
	}
~~~

muka, 2022-01-09:
Great, need to take some time and clean up those wrappers.. Closing, please reopen if needed!

closed 2022-01-09 by muka