muka/go-bluetooth #140: [Bug] Discovery Filtering doesn't work

james-lawrence, 2021-08-12:
the following code doesn't actually filter devices by uuids.

```
	filter := adapter.NewDiscoveryFilter()
	filter.AddUUIDs(uuid)
	discovery, cancel, err := api.Discover(a, &filter)
```

if i put in a junk uuid I expect no devices to be found. (wasn't the case)
if i put in a valid id i expect the devices to be properly filtered. (wasn't the case)

muka, 2021-08-12:
Hi @james-lawrence 
this is a recurring topic and seems to variate based on bluez version. Surely there is a bug, would you like to look into it and  provide a PR eventually?

Thank you

james-lawrence, 2021-08-13:
definitely looks bluez related, at least the proper data is being sent to the service via dbus. thats highly annoying.

for now working around it by manually filtering. do you know where to send bluez bug reports? a cursory search didn't turn up much.

muka, 2021-08-13:
You could try via the bluez mailing list or irc, see http://www.bluez.org/contact/

Let me know, thanks!

muka, 2021-12-27:
Closing, please reopen if you have updates. Thanks

closed 2021-12-27 by muka